FastLED/FastLED #1074: hsv2rgb : Overflow/rollover while scaling up

ocollin, 2020-08-28:
https://github.com/FastLED/FastLED/blob/3a1d880fb7abe5308950358cc1493c5b9e97e3eb/hsv2rgb.cpp#L571

For some RGB values, the scaling up will try to assign values bigger than 255 to r, g or b...
For example, with RGB(218,137,48), after scaling up, r should be 299 but gets assigned to 43 !!

Maybe use uint16_t variables for r, g and b, then let the second scaling (L583) be for any value of total, to compensate for low and too big values
I tried scale down for max(r,g,b)>255 but it breaks the hue value anyway (or my implementation of the scale8, qsub8 are wrong)... I think it is because the rainbow wheel supposes/makes that total is around 255...

I made an .ods file that simulates that (hue calculation was made only for the sample color...)
I also tried to reverse the hsl2rgb function while letting values > 255, and found an almost exact match (depending on how the rounding occurs, and I got the hue manually...)  Dimming back to 255 could be done as the approximation...

[rgb2hsv.zip](https://github.com/FastLED/FastLED/files/5141255/rgb2hsv.zip)

Have fun ;)

kriegsman, 2020-09-04:
Thank you - this is a great ticket report.  I'm going to update the title to reflect that it appears to be a problem with "rgb2hsv_approximate" (not with hsv2rgb), and I'm taking a look at the code now.

ocollin, 2020-09-05:
> Thank you - this is a great ticket report. I'm going to update the title to reflect that it appears to be a problem with "rgb2hsv_approximate" (not with hsv2rgb), and I'm taking a look at the code now.

Sorry for the title... I thought the ticket would be "linked to" the problematic line and thus the function, not that there would be a link to that line...

I am not used to integer math so I don't know how it is supposed to round so the .ods could be a little off...

DrJaymz, 2020-09-12:
rgb2hsv_approximate does not work correctly is this the reason why?
By not properly I mean,  Hues for primary colours are correct but wrong for [largely] secondary colours.  For example, Orange is green.  There's another ticket for that but seems to have been hanging around for years.

ocollin, 2020-09-13:
I think that the overflow/rollover that I mentioned is the (or at least
one) cause of (at least some) problems reported about this function...  I
opened this ticket because I was analyzing the function with the color I
wanted to rgb2hsv and I pinpointed the mentioned line (I am new to GitHub
and I don't really know how reporting bugs are supposed to be done... I
just happened to "fall" on that one...)

I must happen for many (too) bright colors, for witch desaturating leaves a
total value (far) bigger than 255...

As mentioned I made an .ods file that "simulates" the function step by step
(the hue calculation is not implemented for all cases) and reveals that r g
or b should be assigned values bigger than 255...
I also tried another approach that reverses (or tries to) the
hsv2rgb(rainbow) algorithm... that approach shows that having (v) values
bigger than 255 allows to reach a bigger rgb spectrum and gives almost
perfect match (when modifying hsv2rgb to allow those v values).  The
approximation in the rgb2hsv_approximate function could be exactly about
truncating the v value down to 255...

I hope that my reporting will help all those who struggle with this
function...

I know that it is not fun but sometime, when something does not work as
intended it is better to start from scratch... :/

Olivier Collin

zackees, 2026-08-31:
Resolved by merged PR #4086.

closed 2026-08-31 by zackees