FastLED/FastLED #1076: New Feature suggestion 😀 Online Simulator for DemoReel and others

Puneethkumarsn, 2020-09-01:
Hi There

Good day

Thank you so much for the FastLED example libraries. They are indeed very helpful. 🙏
We went ahead with your library and have created a real simulated Arduino which can be compiled and viewed online. the user need not download or install anything. 

You can see it in action here:
https://wokwi.com/arduino/libraries/FastLED/DemoReel100

We will be more than happy to update the simulated page links in the ReadMe file so that the users can really see the library in action!! We can create a PR soon after you inform us.
Please leave a comment or message us.

Here are a few snippets of the online free simulator developed for the Arduino😍 community 😀

 We will be glad to also modify the simulator based on your feedback. Your feedback on the simulator is welcome :) 

[image]

[image]

[image]
[image]

marmilicious, 2020-09-01:
This is so great.  Thank you for sharing this!

Puneethkumarsn, 2020-09-02:
Thank you for the feedback 😀
We will be glad if you have any suggestions to incorporate, or modify.
Kindly let us know.

Shall I add the link in the ReadMe file.. I can create a PR for the same.
Thank you
Puneeth

yesyesuk, 2020-09-02:
This is very cool, thanks for sharing. 
One suggestion I have is to allow changing the geometry of the LED preview box. From a single long LED strip with x LEDs to a matrix of x by y LEDs.

Puneethkumarsn, 2020-09-02:
Definitely. I will come back with the simulation pages for the recommended geometry option as well

Puneethkumarsn, 2020-09-05:
Sorry for the late reply.
I hope this example helps to create a custom matrix. Please give feedback.

https://wokwi.com/arduino/libraries/FastLED/Metaballs
[image]

Thank you
Puneeth

yesyesuk, 2020-09-08:
Hmmm, I was hoping the square box with the LED simulation would change shape depending on the geometry. For example, I would like to see the effect as one long sting of LEDs when I select NUM_ROWS 1 and NUM_COLS 128.

Puneethkumarsn, 2020-09-08:
Please change the row numbers in the .JSON file.. I am on my mobile browser now and I did it as well. It works. 

Please let me know if it works at your end..

yesyesuk, 2020-09-08:
ah, perfect, that is what I was talking about ;) thank you!

Puneethkumarsn, 2020-09-08:
That's great 😀

I would like to rise the PR to add the simulation pages link so that the users of the library can look and also tinker with the display

I will come back with the dedicated simulation pages link for the available examples. Kindly give me some time

Thank you + Best Regards
Puneeth

kostyamat, 2020-11-23:
Some people make matrix-based table lamps and screens.  Could you add a frozen glass (light scattering) effect on top of the matrix?

Puneethkumarsn, 2020-11-28:
sorry for the late reply. I will definitely add it to the list of features request. 
See you in some time

Thanks
Puneeth