FastLED/FastLED #1080: Led0 flickers w/ esp8266, pacifica example, and wifi

d-a-v, 2020-09-13:
Whether interrupts are enabled or not, led0 flickers when WiFi is enabled.
I've read #504 and users seems to have no issue with esp8266 and WiFi.
Pacifica works well when WiFi is disabled. I have no level converter.
Here's the minimum changes to make Pacifica example flicker.

Led0 specifically flickers at least when esp8266 receives WiFi packets (tried with `ping` and `nmap` but there's "natural" trafic too).

```
diff --git a/examples/Pacifica/Pacifica.ino b/examples/Pacifica/Pacifica.ino
index 29ac9fc..7da3607 100644
--- a/examples/Pacifica/Pacifica.ino
+++ b/examples/Pacifica/Pacifica.ino
@@ -5,14 +5,17 @@
 //  For Dan.
 //
 
-#define FASTLED_ALLOW_INTERRUPTS 0
+#include <ESP8266WiFi.h>
+
+#define FASTLED_ALLOW_INTERRUPTS 1
+#define INTERRUPT_THRESHOLD 0
 #include <FastLED.h>
 FASTLED_USING_NAMESPACE
 
-#define DATA_PIN            3
+#define DATA_PIN            2
 #define NUM_LEDS            60
 #define MAX_POWER_MILLIAMPS 500
-#define LED_TYPE            WS2812B
+#define LED_TYPE            WS2813
 #define COLOR_ORDER         GRB
 
 //////////////////////////////////////////////////////////////////////////
@@ -20,6 +23,8 @@ FASTLED_USING_NAMESPACE
 CRGB leds[NUM_LEDS];
 
 void setup() {
+WiFi.mode(WIFI_STA);
+WiFi.begin("ssid", "psk");
   delay( 3000); // 3 second delay for boot recovery, and a moment of silence
   FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS)
         .setCorrection( TypicalLEDStrip );
```

d-a-v, 2020-09-14:
FYI, when using another random WS2812 library, there's no issue when WiFi is enabled and shaken on the same hardware (still without level shifter).
However, your API is very nice and fluent to use :]

Is there anything I could further do to help ?

CRCinAU, 2020-10-08:
@d-a-v - I've been suffering with this for a LONG, long time....

I did find a branch that used the RX pin on, say, the D1 Mini and used DMA to output directly to the LED strip, but after updating platformio, this no longer compiles.

I'm pretty sure its this one: https://github.com/EmteZogaf/FastLED/tree/dma

However - it is quite a bit behind, and with the changes for ISO C++17, there's a LOT of errors around the use of register that I tried to fix and just ended up breaking it more...

Even though we're stuck using the RX pin, its normally smooth sailing...

It would be nice to see this fixed up and end up in the proper FastLED though... I tried to advocate for this a long time ago with no success...

CRCinAU, 2020-10-08:
For what its worth, my version of it is here: https://git.crc.id.au/netwiz/ESP8266_Code/src/branch/master/lib/FastLED_DMA

CRCinAU, 2020-10-10:
For what its worth - this crash is NOT related to my changes to the FastLED DMA code as I thought it was...

I reverted to the ESP8266 Android v2.7.3 release and everything works perfectly - and at some point, it worked perfectly on v3.0.0 too. As such, its likely a regression in stuff that *isn't* my changes... It also means that if someone was able to bring my changes into alignment with FastLED here, it could be merged...

I'd do it myself - but with my workload right now, I'm not sure thats very realistic...

DrCos, 2021-06-20:
Having this problem with LED0 - Wemos D1 (8266) and FastLED.
If I download with Arduino and FastLED library 3.3.0 - no flicker.
Trying to use PlatformIO and several FastLED libraries 3.3.x to 3.4 - LED0 flickers.  On two different boards, and two different strips.
I have a data line resistor, and the ground is ground for the board and strip.  Using GPI0 (D3) for data.

Any suggestions?

closed 2024-10-12 by zackees