FastLED/FastLED #1081: Big fan of the lib having problem with ESP32, first led is going crazy.

sblantipodi, 2020-09-15:
Hi,
I'm big fan of Fastled.

Using it on ESP8266 for various "applications".

Now I'm trying the same code on ESP32 using the usual diagram:
https://github.com/sblantipodi/glow_worm_luciferin/blob/master/assets/img/ambilight_bb.png

but the first led does not respond correctly. all other leds works correctly but first led not.
sometimes gets the right command, sometimes is completely unresponsive and stay all white or all green, or all yellow.

using a logic level converter and a 220ohm resistance, tried even a 470 resistance by changing the LEDs on every try but first led never works ok.

Any idea? Is this a software problem?

Hackswell, 2020-09-16:
May not mean anything, but on the level shifter, common ground should be shared between both LV and HV.  The diagram is missing LV grounding.  When using a level shifter, you may not need a resistor at all.  Same goes for the capacitor...I think it's just for voltage smoothing?

Usually on WS2812s, when ONLY the first LED is wonky is when it's not getting a proper 5V signal in, and IT is acting as the level booster. 

Have you measured your 5V in?  Is it actually 5V and not like 4V or something?  Just something else to check.

sblantipodi, 2020-09-16:
Thanks for the answer @Hackswell I appreciate it.
I added the ground to the lv on level shifter as per your suggestion but it doesn't helped.
I use the resistance just to protect the digital pin a bit, I tried with and without the resistance but it doesn't helped. 

Caps is there for just voltage smoothing even if I have a good bft lighting power supply 5V/8A just for five LEDs. 

5v measures 5.5V with my voltage meter, this power supply is supposed to be made specifically for ws2812 strip.

sblantipodi, 2020-09-16:
My software is configured to light up up to 100leds and I have only five connected but I don't think that this can be a problem

sblantipodi, 2020-09-16:
tried same software, same circuit on the breadboard with the same power supply, 
ESP8266 works well, 
ESP32 have first led STUCK.

is this a known problem on FastLED?

VinceW31, 2020-10-04:
Ive got similar issues, try this:

Connect all 100 LEDs and try out your code on the ESP32 to see if its just the first LED.  On my strip its not just the first led but any LED after 80 is playing up too, same code works fine on Arduino MEGA.

sblantipodi, 2020-10-04:
ok it seems to be proved that Fastled does not like ESP32

sblantipodi, 2020-10-20:
is there someone whith this problem?

do you find some errors in this wiring diagrams assumed that the microcontroller in use is a 
Lolin D32 running ESP32?

sblantipodi, 2020-12-24:
I solved by simply removing this from my  setup()
`digitalWrite(LED_BUILTIN, HIGH);`

closed 2020-12-24 by sblantipodi