FastLED/FastLED #1085: Not working on DFRobot Beetle ESP32

hobzcalvin, 2020-09-24:
I can't get FastLED to work with the [Beetle ESP32](https://wiki.dfrobot.com/Beetle_ESP32_SKU_DFR0575). I've tried several combinations of I/O pins, supported LED chipsets (APA102, WS2812B), #define FASTLED_FORCE_SOFTWARE_PINS on/off, etc. It works fine on other ESP32 dev boards. Signal output from the pins looks the same as a working board on an oscilloscope. I suspected 3.3V pin output could be the problem, but other ESP32 boards at this pin voltage work fine.

sblantipodi, 2020-10-01:
Using Lolin ESP32 here and the first led never works well, others LEDs works well
Is this a problem of the microcontroller or a FastLED one?

hobzcalvin, 2020-10-01:
I suspect the microcontroller, or more likely its custom Arduino “board”
code which is needed for proper pinouts etc, is doing something strange. I
can’t get anything to work on the board with the generic ESP32 Dev Board
profile.

Grant Patterson
http://revoltlabs.co/portfolio
http://instagram.com/grantpatterson
http://www.etsy.com/shop/ReVoltLabs

sblantipodi, 2020-10-03:
@hobzcalvin if you find a solution please update this issue and let us know, I will do the same.

samguyer, 2020-12-30:
@hobzcalvin Did you resolve this issue? If not, I can take a look. I suspect that it has to do with the way the pins are named on that board.

hobzcalvin, 2020-12-30:
I did not; that would be awesome. Yeah, I had some issues getting things to
work with the strangely-named pins. IIRC there were two different Beetle
boards I could choose from in Arduino and only one had the D# naming scheme
(and the other one didn't work at all?) Thanks!!