FastLED/FastLED #1088: Send FastLED info through Serial.

Quantum99k, 2020-09-26:
How can I send FastLED information through serial so i can have another Arduino control the led.
I have a library (FHT.h) that isn't compatible with the with the esp8266 so its uploaded on an arduino micro and i want to transmit that data to the esp8266 running the fastled webserver

marmilicious, 2020-10-14:
@henrygab Do I remember something about you working on an alternate streaming output type, instead of an LED_TYPE such as APA102 or WS2812?

@Quantum99k  The CRGB leds array is simply an array of R,G,B data, so you should be able to make your own custom "show()" function that transmits the data out however you'd like.

henrygab, 2020-10-14:
@marmilicious Hi Marc, I don't believe that was me.  Sorry!

Quantum99k, 2020-10-15:
@marmilicious I've never modified a library before. And one as big as this one would be difficult since i don't know any about the the internals such as naming schema of this library. I have looked through it but i dont know what im looking at quit yet.I did try to serial.print the pwm pin but that didn't do anything.

5chmidti, 2020-10-15:
I don't know how one connects two mcu's via serial, but this should work:
instead of `FastLED.show();` write `Serial.write(leds,NUM_LEDS*sizeof(CRGB));`, maybe inside of an `EVERY_N_MILLIS`.
And to receive `Serial.readBytes(leds,NUM_LEDS*sizeof(CRGB));`.
From Arduino [docs on Serial](https://www.arduino.cc/reference/en/language/functions/communication/serial/)
Note: untested, never done this, no idea if there is a better way.

Quantum99k, 2020-10-15:
Thanks @5chmidti . I will try it tonight an let you know

marmilicious, 2020-10-16:
@Quantum99k there's no need to modify the library.  Where you would normally call show() to display your pixels, instead call a custom function you create.  Name it sendLedData() or showSerial() or whatever makes sense to you.  This function will send the data out using serial to the other controller.

Since we use github "issues" here mostly just for bugs or improvements in the library itself, for general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED

zackees, 2024-10-12:
This is trivial to implement in user land so not going to implement it in core driver land.

closed 2024-10-12 by zackees