FastLED/FastLED #1090: AttinyCore, FastLED and ColorPalette Error

link0687, 2020-09-27:
Hi,
I am using AttinyCore (using an attiny2313a) and FastLED library. Once I've added the blank sketch with the FastLED library included it will compile fine with no errors. As soon as I add the "ColourPalette" example from FastLED library I get a compiler error. I have attached this below in a text file. If you can help me trouble shoot this or point me in the write direction that would be great. I have labeled in the text file where a compiler error begins by using "Orange Error Writing". Please excuse the crudeness
[Error COde.txt](https://github.com/FastLED/FastLED/files/5287811/Error.COde.txt)
. Thanks!

kostyamat, 2020-10-23:
This is not an error, just a notification.

marmilicious, 2020-10-25:
@link0687  Did you get the ColorPalette to work?

link0687, 2020-10-26:
@marmilicious i did but not with the attiny2313. I redesigned the PCB to incorporate the attiny167 and it seems to be working good now.

marmilicious, 2020-10-26:
@link0687  Glad you have something working.  Please share your project with the reddit group at some point if you can.  
Feel free to close this ticket if it's no longer needed.