FastLED/FastLED #1092: showColor on TeensyLC doesn't show the color

pictographer, 2020-09-29:
Was wondering if
```
FastLED.showColor(CRGB::Green);
FastLED.show();
```
would be equivalent to
```
fill_solid(leds, NUM_LEDS, CRGB::Green);
FastLED.show();
```
The `fill_solid` works as expected. The `showColor` doesn't appear to change the state of the LEDs.

Also wondering if `FastLED.show()` should be redundant with `FastLED.showColor(...)`.

I'm using a TeensyLC and FastLED version 3.003.001.

pictographer, 2020-09-30:
I should add, the LEDs are 8 APA102.

marmilicious, 2020-10-15:
I don't have a TeensyLC but I tested with a Teensy3.2 and LPD8806 pixels.

Including the show() line after the showColor() line caused basically a black display with a flickering of the color.  This isn't what I expected.  Commenting out the show() line like this did correctly display the color.
`
void loop() {
  fill_solid(leds, NUM_LEDS, CRGB::Green);
  //FastLED.show();
}
`

Doing this caused the display to turn on/off every 500ms.
`
  fill_solid(leds, NUM_LEDS, CRGB::Green);
  delay(500);
  FastLED.show();
  delay(500);
`

One more test:
`
  FastLED.showColor(CRGB::Green, 255);
  Serial.print("\n Using showColor -- leds[0]: ");
  Serial.print(leds[0].r); Serial.print(" ");
  Serial.print(leds[0].g); Serial.print(" ");
  Serial.print(leds[0].b);
  delay(2000);

  fill_solid(leds, NUM_LEDS, CHSV(random8(),255,255));
  FastLED.show(16);
  Serial.print("\n Using show() -- leds[0]: ");
  Serial.print(leds[0].r); Serial.print(" ");
  Serial.print(leds[0].g); Serial.print(" ");
  Serial.print(leds[0].b);
  delay(2000);
`

After observing this last test it appears that using showColor doesn't put the color data into the CRGB array, but rather pushes it straight out to the pixels.  Looking at the serial monitor, only the r,g,b data for the fill_solid color is printed out, and we never see 0,255,0 for green as I would have guessed.  

So to answer your original question @pictographer , no, they don't seem to be equivalent!  Data seems to be pushed straight out to the pixels without being put into the CRGB array, and I would guess this is by design.  Hopefully we can get some history and info about that from @kriegsman 

I had never used showColor before as I've always used fill_solid and then called show().  Personally I think setting the color with fill_solid and then calling the show() line makes it more clear how some code is working, but there might indeed be useful cases where not changing the current r,g,b data in the CRGB array is useful.