FastLED/FastLED #1097: Flickering when use ws2812B panel version with ESP32

DJDOUBLEG, 2020-10-14:
I maked WordkClock with ws2812b stip. Work well. 
I change the strip to panel led.
[image]
And have this issue:
https://photos.app.goo.gl/W9MqDcL9iQSmjSbQ6
Same processzor same code.

marmilicious, 2020-10-15:
It's possible you might need to add a level shifter such as SN74HCT245N or SN74HCT125N if you're not already using one.  Since this is not a library issue, please ask your question on the reddit group:  https://www.reddit.com/r/FastLED  
We use github "issues" here mostly just for bugs or improvements in the library itself.

BatsIhor, 2020-10-16:
Hi, I researched this issue quite a bit and it is library issue, you can partially reduce flickering my moving FastLED to separate core, but still when I use wifi it's flickering a bit.

DJDOUBLEG, 2020-10-16:
Hi!

I found the cause of the error. 5V was too much. The voltage had to be set
to 4.8v. now there is no problem.
I don't understand why..

Ihor Bats <notifications@github.com> ezt írta (időpont: 2020. okt. 16., P
22:53):

marmilicious, 2020-10-17:
@DJDOUBLEG This is because the ESP32 is 3.3V and the LEDs are 5V.  By lowering the voltage to 4.8V you got the logic level within tolerance to work.  Using a level shifter as I mentioned above would have solved this without needed the reduce your voltage to the strips, as it would bump the data signal up from 3.3V to 5V before going to the LEDs.  

The fact that that sending the 3.3V signal the LEDs worked with your first setup is just luck.  Some times the voltage levels are just close enough that it will work.  Some times it will work intermittently.  Sometimes they won't at all.  When you switched to different LEDs it was just different enough that it didn't work.  If you want true reliability use a level shifter when using a 3.3V controller so you have a guaranteed 5V data signal going into the LEDs.

Here are some links if you want to learn more about logic levels:
https://www.allaboutcircuits.com/textbook/digital/chpt-3/logic-signal-voltage-levels/

https://www.instructables.com/A-Quick-Guide-on-Logic-Level-Shifting/

https://learn.sparkfun.com/tutorials/logic-levels/all#:~:text=Put%20simply%2C%20a%20logic%20level,signals%20HIGH%20or%20LOW%2C%20respectively.

Feel free to close the this issue if you consider it solved.

DJDOUBLEG, 2020-10-17:
No. The mistake is not here! Because I tried it with arduino mega. The data
output is 5V, if I give him 5V with the Dcdc converter, he's made the same
mistake. If it was only 4.8V, it went perfectly.
That's what you're trying to explain.

Marc Miller <notifications@github.com> ezt írta (időpont: 2020. okt. 17.,
Szo 4:52):

closed 2020-10-21 by DJDOUBLEG