FastLED/FastLED #1102: Multiple strips - different types

tomtomhoward, 2020-10-25:
Hi,

I am having a little trouble but just want to confirm it's me.
I am trying to use two different types of strip off the same controller, WS2811 on one pin and Neopixel on another. It's because the WS2811 is a fairly light string and the Neopixel/WS2812 is tape.

It's a halloween display.

So my declaring code is basically 

`#define DATA_PIN_WINDOW 27  //Define led data pin for window string
#define LED_TYPE_WINDOW WS2811  //define type of led window
#define NUM_LEDS_WINDOW 50  //num of leds in window string
#define DATA_PIN_PUMPKIN 26  //Define led data pin for pumpkin string
#define LED_TYPE_PUMPKIN WS2811  //define type of led pumpkin
#define NUM_LEDS_PUMPKIN 45  //num of leds in pumpkin strip`

The main reason for declaring two types is the WS2811 is RGB colour order and the WS2812 is GRB.
Declaring two types makes spurious colours in the first few LEDs across both strips. If I use either type for both both strings are stable but colour order is wrong.

I'm assuming it's something to do with the timer or similar, although they use the same clock freq?
I can just declare my colours manually in one and switch the order around to fix but didn't know if it should count as a bug, I couldn't find any documentation relating to different strip types.

tomtomhoward, 2020-10-25:
So the behaviour seems to be...

On the Neopixel strip, the first two pixels only are correct then everything else R and G is swapped
On the WS2811 strip, the first two pixels only are swapped and the rest of the string is correct

tomtomhoward, 2020-10-25:
So.... I changed my Neopixels for more WS2811 but found a different error... the fairly lights (window) are RGB and my strip (pumpkins) is BRG

Using the following code to try sort out the colours when addleds:

`FastLED.addLeds<LED_TYPE_PUMPKIN, DATA_PIN_PUMPKIN, BRG>(pumpkin, NUM_LEDS_PUMPKIN); //add LEDs
FastLED.addLeds<LED_TYPE_WINDOW, DATA_PIN_WINDOW, RGB>(window, NUM_LEDS_WINDOW); //add LEDs`

Both strips follow whichever command is run first, so BRG if Pumpkin is declared first or RGB if Window is. I don't seem to be able to declare them separately.

marmilicious, 2020-10-25:
Does it sound like this one?  Also, what MCU are you using?
https://github.com/FastLED/FastLED/issues/827#issuecomment-505668673

tomtomhoward, 2020-10-25:
That does sound like the RGB order, but the initial problem is outside of that. Using the multiple instances with 2 different LED types seems to cause problems in the colour order (or another corruption) in the first couple of units of each string.
For the time being I've just hard coded some non CRGB colours and just switch the colours when they are called on one of the strips, and not use the library colours.

tomtomhoward, 2020-10-25:
I'm using ESP32 (WROOM 32)