FastLED/FastLED #1110: Multiple Controller wiki page name

costyn, 2020-11-10:
The Multiple Controller Github wiki page should be called Multiple Strips? I thought it was called this before; it's become a lot harder to find now through Google.

https://github.com/FastLED/FastLED/wiki/Multiple-Controller-Examples

marmilicious, 2020-11-10:
I don't think the name has changed, at least not anytime in the past 6 years.  
https://github.com/FastLED/FastLED/wiki/Multiple-Controller-Examples/_history

costyn, 2020-11-10:
Ok, then it's just my terrible memory. 😁 But, don't you agree the name is confusing?

marmilicious, 2020-11-10:
"Controller" refers to the various Controller classes in the library.  While the page name is perhaps a bit technical, it is being accurate.  If you are setting up multiple strips then you are indeed using multiple pixel Controllers.  :)