FastLED/FastLED #1111: Bluetooth connection nRF52 don't work when I have 4 RGB leds strip initialized

Strangistra, 2020-11-11:
Hello

I'm Strangistra and I work on a Adafruit Feather nrf52832.

I have a strange behavior with the Bluetooth connection.

I have a code to drive 4 RGB leds strips. For this I used the FastLed library and the MultiArray exemple.
`
#include <FastLED.h>

#define NUM_LEDS_PER_STRIP 12
CRGB redLeds[NUM_LEDS_PER_STRIP];
CRGB greenLeds[NUM_LEDS_PER_STRIP];
CRGB blueLeds[NUM_LEDS_PER_STRIP];

// For mirroring strips, all the "special" stuff happens just in setup.  We
// just addLeds multiple times, once for each strip
void setup() {

  Serial.begin(115200);
 
  FastLED.addLeds<SK6812,A0,GRB>(redLeds,NUM_LEDS_PER_STRIP);

  FastLED.addLeds<SK6812,A1,GRB>(greenLeds, NUM_LEDS_PER_STRIP);

  FastLED.addLeds<SK6812,A2,GRB>(blueLeds, NUM_LEDS_PER_STRIP);

  FastLED.addLeds<SK6812,A3,GRB>(blueLeds, NUM_LEDS_PER_STRIP);
}

void loop() {
  for(int i = 0; i < NUM_LEDS_PER_STRIP; i++) {
    // set our current dot to red, green, and blue
    redLeds[i] = CRGB::Red;
    greenLeds[i] = CRGB::Green;
    blueLeds[i] = CRGB::Blue;
    FastLED.show();
    // clear our current dot before we move on
    redLeds[i] = CRGB::Black;
    greenLeds[i] = CRGB::Black;
    blueLeds[i] = CRGB::Black;
    delay(100);
  }

  for(int i = NUM_LEDS_PER_STRIP-1; i >= 0; i--) {
    // set our current dot to red, green, and blue
    redLeds[i] = CRGB::Red;
    greenLeds[i] = CRGB::Green;
    blueLeds[i] = CRGB::Blue;
    FastLED.show();
    // clear our current dot before we move on
    redLeds[i] = CRGB::Black;
    greenLeds[i] = CRGB::Black;
    blueLeds[i] = CRGB::Black;
    delay(100);
  }
}
`
This code work correcly.

Now I add this code on the Adafruit Dual_Bleuart exemple :
`
#include <bluefruit.h>
#include <FastLED.h>
#define NUM_LEDS_PER_STRIP 12

CRGB redLeds[NUM_LEDS_PER_STRIP];
CRGB greenLeds[NUM_LEDS_PER_STRIP];
CRGB blueLeds[NUM_LEDS_PER_STRIP];

[... 505 lines cut ...]
 * Callback invoked when uart received data
 * @param cent_uart Reference object to the service where the data
 * arrived. In this example it is clientUart
 */
void cent_bleuart_rx_callback(BLEClientUart& cent_uart)
{
  char str[20+1] = { 0 };
  cent_uart.read(str, 20);
     
  Serial.print("[Cent] RX: ");
  Serial.println(str);

  if ( bleuart.notifyEnabled() )
  {
    // Forward data from our peripheral to Mobile
    bleuart.print( str );
  }else
  {
    // response with no prph message
    clientUart.println("[Cent] Peripheral role not connected");
  } 
}
`

Have you already see this strange issue?

mklemarczyk, 2022-08-07:
The BLE libraries does not cooperate with the FastLED. I have the same issue with Arduino 33 IoT.

Ref article
https://www.davidbritch.com/2021/08/fastled-and-arduinoble.html

zackees, 2022-08-07:
From your code I see that you are using SK6812, which is a clone of the WS2812 aka Neopixel.

I would suggest you try this library over here:
https://github.com/adafruit/Adafruit_NeoPixel

Or try turning off hardware spi:
```C++
#define FASTLED_FORCE_SOFTWARE_SPI
#include <FastLED.h>
```

topsub, 2024-11-02:
Any updates on support for nRF52? Built a customer pcb board not realizing there isn't support for this chip

zackees, 2024-11-12:
> Any updates on support for nRF52? Built a customer pcb board not realizing there isn't support for this chip

I've worked with many of these boards. for some reason there's a bunch of defines in fast led to specific nr52 boards. You may get lucky by just using the closest board in this nr52 family