FastLED/FastLED #1114: inoise8(i) returns a constant value of 128 from i = 3061 to 3599

marmilicious, 2020-11-15:
This seems to be a strange one, reported by djbog (Scott Marley, @s-marley ) in the reddit group.  I ran my own test (on an UNO) and got the same results.

When running inoise8(i) it returns a constant value of 128 from the range of i = 3061 to 3599.  

```
for(uint16_t i = 0; i < 5000; i++) {
  Serial.print(i);
  Serial.print(", \t");
  Serial.println(inoise8(i));
}
```

Here is Scott's reddit post with a plot that shows the "flat" in the otherwise nice curvey output.  
https://www.reddit.com/r/FastLED/comments/ju7ams/something_i_dont_understand_with_inoise8x_output/

Any ideas why it would do this?

[image]

kriegsman, 2020-11-24:
Hrm.  Interesting.  Thank you for the issue report here and the _really_ useful graph.

I'll assign this to myself and queue it up to take a look.  I suspect it's a loss-of-precision issue somewhere in the math.

devoh747, 2020-12-01:
Do we know if both examples were run on the same processor/architecture?

devoh747, 2020-12-02:
[image]

Looks to be the same on ESP32 using arduino 1.8.13 and FaastLED 3.3.3

closed 2026-09-12 by zackees