FastLED/FastLED #1115: ESP32 Reboots with webserver if fastled library is included.

zekageri, 2020-11-19:
**I'm using ESPAsyncwebserver.**

### Here is the backtrace of the reboot:

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x4013c000  PS      : 0x00060034  A0      : 0x800811c0  A1      : 0x3ffbe780    
A2      : 0x00000000  A3      : 0x3ffc117c  A4      : 0x5d5ee776  A5      : 0x3ffd0f70    
A6      : 0x00000000  A7      : 0x3ffd0fac  A8      : 0x8008111e  A9      : 0x3ffbe760    
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x0ffd114c  A13     : 0x00000000    
A14     : 0x00000080  A15     : 0x00000000  SAR     : 0x0000000e  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x40081060  LEND    : 0x40081067  LCOUNT  : 0x00000000
Core 1 was running in ISR context:
EPC1    : 0x400876b9  EPC2    : 0x00000000  EPC3    : 0x4009318a  EPC4    : 0x4013c000

Backtrace: 0x4013c000:0x3ffbe780 0x400811bd:0x3ffbe7b0 0x400853cd:0x3ffbe7d0 0x40093187:0x3ffd0fd0 0x40093a11:0x3ffd0ff0 0x40087cf0:0x3ffd1010 0x4013dfdf:0x3ffd1080 0x400e292c:0x3ffd10b0 0x400df571:0x3ffd10e0 0x400dfcbf:0x3ffd1110 0x400dfd9b:0x3ffd1170 0x400dff4e:0x3ffd11a0 0x400e09b9:0x3ffd11d0 0x400deadd:0x3ffd1220 0x4010f75d:0x3ffd1350 0x4000bcc5:0x3ffd1370 0x40152ed5:0x3ffd1390 0x401458c3:0x3ffd13b0 0x40145df7:0x3ffd13e0 0x401460dd:0x3ffd1420 0x400d51fc:0x3ffd14a0 0x400d5223:0x3ffd14d0 0x400dbe87:0x3ffd14f0 0x400d9a7a:0x3ffd15a0 0x400d2876:0x3ffd15e0 0x400dc321:0x3ffd1630 0x400d9c5d:0x3ffd1680 0x400d9d21:0x3ffd16c0 0x400d9f89:0x3ffd1700 0x40144ad9:0x3ffd1720 0x40144b55:0x3ffd1760 0x401451fe:0x3ffd1780 0x40089689:0x3ffd17b0

```

### Decoded:

```
#0  0x4013c000:0x3ffbe780 in rmt_set_tx_intr_en at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt.c:366
  #1  0x400811bd:0x3ffbe7b0 in ESP32RMTController::interruptHandler(void*) at lib\FastLED\src\platforms\esp\32/clockless_rmt_esp32.cpp:417
  #2  0x400853cd:0x3ffbe7d0 in _xt_medint3 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1323
  #3  0x40093187:0x3ffd0fd0 in esp_rom_spiflash_read_data at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:413
  #4  0x40093a11:0x3ffd0ff0 in esp_rom_spiflash_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:587
  #5  0x40087cf0:0x3ffd1010 in spi_flash_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:567
  #6  0x4013dfdf:0x3ffd1080 in esp_partition_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/partition.c:242
  #7  0x400e292c:0x3ffd10b0 in littlefs_api_read at lib\LITTLEFS\src/littlefs_api.c:22
  #8  0x400df571:0x3ffd10e0 in lfs_bd_read at lib\LITTLEFS\src/lfs.c:3459
  #9  0x400dfcbf:0x3ffd1110 in lfs_dir_getslice at lib\LITTLEFS\src/lfs.c:3459
  #10 0x400dfd9b:0x3ffd1170 in lfs_dir_get at lib\LITTLEFS\src/lfs.c:3459
  #11 0x400dff4e:0x3ffd11a0 in lfs_dir_getinfo at lib\LITTLEFS\src/lfs.c:3459
  #12 0x400e09b9:0x3ffd11d0 in lfs_stat at lib\LITTLEFS\src/lfs.c:3459
  #13 0x400deadd:0x3ffd1220 in vfs_littlefs_stat at lib\LITTLEFS\src/esp_littlefs.c:231
  #14 0x4010f75d:0x3ffd1350 in esp_vfs_stat at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:972
  #15 0x4000bcc5:0x3ffd1370 in ?? ??:0
  #16 0x40152ed5:0x3ffd1390 in stat at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/syscalls/../../../.././newlib/libc/syscalls/sysstat.c:12
  #17 0x401458c3:0x3ffd13b0 in VFSFileImpl::VFSFileImpl(VFSImpl*, char const*, char const*) at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:127 (discriminator 1)
  #18 0x40145df7:0x3ffd13e0 in void __gnu_cxx::new_allocator<VFSFileImpl>::construct<VFSFileImpl, 
VFSImpl*&, char const*, char const*&>(VFSFileImpl*, VFSImpl*&, char const*&&, char const*&) at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:127 (discriminator 1)
      (inlined by) std::enable_if<std::__and_<std::allocator_traits<std::allocator<VFSFileImpl> >::__construct_helper<VFSFileImpl, VFSImpl*&, char const*, char const*&>::type>::value, void>::type 
std::allocator_traits<std::allocator<VFSFileImpl> >::_S_construct<VFSFileImpl, VFSImpl*&, char const*, char const*&>(std::allocator<VFSFileImpl>&, VFSFileImpl*, VFSImpl*&, char const*&&, char const*&) at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/alloc_traits.h:256 (discriminator 1)
      (inlined by) decltype (_S_construct({parm#1}, {parm#2}, (forward<VFSImpl*&>)({parm#3}), (forward<char const*>)({parm#3}), (forward<char const*&>)({parm#3}))) std::allocator_traits<std::allocator<VFSFileImpl> >::construct<VFSFileImpl, VFSImpl*&, char const*, char const*&>(std::allocator<VFSFileImpl>&, VFSFileImpl*, VFSImpl*&, char const*&&, char const*&) at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/alloc_traits.h:402 (discriminator 1)
      (inlined by) std::__shared_ptr<VFSFileImpl, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<VFSFileImpl>, VFSImpl*&, char const*, char const*&>(std::_Sp_make_shared_tag, std::allocator<VFSFileImpl> const&, VFSImpl*&, char const*&&, char const*&) at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:1129 (discriminator 1)
  #19 0x401460dd:0x3ffd1420 in std::shared_ptr<VFSFileImpl>::shared_ptr<std::allocator<VFSFileImpl>, VFSImpl*&, char const*, char const*&>(std::_Sp_make_shared_tag, std::allocator<VFSFileImpl> const&, VFSImpl*&, char const*&&, char const*&) at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:127 (discriminator 1)
      (inlined by) std::shared_ptr<VFSFileImpl> std::allocate_shared<VFSFileImpl, std::allocator<VFSFileImpl>, VFSImpl*&, char const*, char const*&>(std::allocator<VFSFileImpl> const&, VFSImpl*&, 
char const*&&, char const*&) at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr.h:614 (discriminator 1)
      (inlined by) std::shared_ptr<VFSFileImpl> std::make_shared<VFSFileImpl, VFSImpl*&, char const*, char const*&>(VFSImpl*&, char const*&&, char const*&) at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr.h:630 (discriminator 1)  
      (inlined by) VFSImpl::open(char const*, char const*) at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\libraries\FS\src/vfs_api.cpp:44 (discriminator 1)
  #20 0x400d51fc:0x3ffd14a0 in fs::FS::open(char const*, char const*) at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\libraries\FS\src/FS.cpp:269
  #21 0x400d5223:0x3ffd14d0 in fs::FS::open(String const&, char const*) at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\libraries\FS\src/FS.cpp:269
  #22 0x400dbe87:0x3ffd14f0 in AsyncFileResponse::AsyncFileResponse(fs::FS&, String const&, String const&, bool, std::function<String (String const&)>) at lib\ESPAsyncWebServer\src/WebResponses.cpp:699
  #23 0x400d9a7a:0x3ffd15a0 in AsyncWebServerRequest::beginResponse(fs::FS&, String const&, String const&, bool, std::function<String (String const&)>) at lib\ESPAsyncWebServer\src/WebRequest.cpp:797
  #24 0x400d2876:0x3ffd15e0 in std::_Function_handler<void (AsyncWebServerRequest*), HTTP_Requests()::{lambda(AsyncWebServerRequest*)#2}>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) at lib/FastLED/src/pixeltypes.h:459
      (inlined by) _M_invoke at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #25 0x400dc321:0x3ffd1630 in std::function<void (AsyncWebServerRequest*)>::operator()(AsyncWebServerRequest*) const at lib\ESPAsyncWebServer\src/StringArray.h:73
      (inlined by) AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at lib\ESPAsyncWebServer\src/WebHandlerImpl.h:132
  #26 0x400d9c5d:0x3ffd1680 in AsyncWebServerRequest::_parseLine() at lib\ESPAsyncWebServer\src/WebRequest.cpp:797
[... cut ...]
        }else if(ANIMATION->TYPES[5]){
          rainbowWithGlitter();
        }else if(ANIMATION->TYPES[6]){
          juggle();
        }else if(ANIMATION->TYPES[7]){
          water();
        }else if(ANIMATION->TYPES[8]){
          fire();
        }else if(ANIMATION->TYPES[9]){
          pride();
        }else if(ANIMATION->TYPES[10]){
          sinelon();
        }else if(ANIMATION->TYPES[11]){
          bpm();
        }
        FastLED.show();
        vTaskDelay(1000 / FRAMES_PER_SECOND);
      }
    }
  }
```

samguyer, 2020-11-25:
@zekageri Are you accessing flash memory in this application? There are some tricky interactions between flash access and other code on the ESP32. On thing you can try: add the following line *before* including fastled.h:

#define FASTLED_ESP32_FLASH_LOCK 1

zekageri, 2020-11-25:
Iam. I use LittleFS to serve the webpages, writing error log, etc. I will try that thanks.

samguyer, 2020-11-25:
@zekageri Let me know if that works.

zekageri, 2020-11-26:
Nop, didnt work.

zekageri, 2020-11-26:
I have got a half way working workaround. In the "/" handler i suspend the led animation task and after 2 sec i resume in it.

```
void handleRequest(AsyncWebServerRequest *request) {
    vTaskSuspend(Animation_Task_Handle);
    Can_Resume_Animations = true;
    Start_AnimationTime = millis();
    AsyncWebServerResponse* response = request->beginResponse(LITTLEFS, "/LampBundle.html.gz", "text/html");
    response->addHeader("Content-Encoding", "gzip");
    request->send(response);
 }
```

```
void loop() {
  if(Can_Resume_Animations && ( (millis() - Start_AnimationTime) >= 2000) ){
    Can_Resume_Animations = false;
    vTaskResume(Animation_Task_Handle);
  }
  vTaskDelay(10);
}
```

This way working roughly half of the time. :'D

HamzaHajeir, 2020-11-29:
I've pretty the same issue : 

I'm using [H4Plugins ](https://github.com/philbowles/h4plugins/)Framework which uses ESPAsyncWebServer. 

Some times I got this crash just after h4setup() *Which is "user" setup()*

Decoded stack trace : 

```
Lift off h4setup()
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x40150098  PS      : 0x00060034  A0      : 0x800811c4  A1      : 0x3ffbe780
A2      : 0x00000000  A3      : 0x3ffc11f0  A4      : 0x15b3d314  A5      : 0x00000000
A6      : 0x3ffc27d0  A7      : 0xffffffff  A8      : 0x80081122  A9      : 0x3ffbe760
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x0ffd114c  A13     : 0x00000000
A14     : 0x00000080  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x40081064  LEND    : 0x4008106b  LCOUNT  : 0x00000000
Core 1 was running in ISR context:
EPC1    : 0x40089469  EPC2    : 0x00000000  EPC3    : 0x40087560  EPC4    : 0x40150098

Backtrace: 0x40150098:0x3ffbe780 0x400811c1:0x3ffbe7b0 0x400853f5:0x3ffbe7d0 0x4008755d:0x3ffba300 0x4008365f:0x3ffba320 0x400896b1:0x3ffba340
  #0  0x40150098:0x3ffbe780 in rmt_tx_stop at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt.c:363
  #1  0x400811c1:0x3ffbe7b0 in ESP32RMTController::interruptHandler(void*) at .pio\libdeps\esp32doit-devkit-v1\FastLED\src\platforms\esp\32/clockless_rmt_esp32.cpp:417
  #2  0x400853f5:0x3ffbe7d0 in _xt_medint3 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1323
  #3  0x4008755d:0x3ffba300 in spi_flash_op_block_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:82 (discriminator 1)
  #4  0x4008365f:0x3ffba320 in ipc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:62
  #5  0x400896b1:0x3ffba340 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

```

What I found is that the interrupt handler had looked out for ESP32RMTController::interruptHandler in IRAM, But it wasn't set yet in IRAM at that moment. 

however I sometimes get this crash once at the beginning, Twice, Multiple. But then it boots successfully. 

But anyway that's annoying and needs a real fix.

What could we do ?

samguyer, 2020-11-29:
@HamzaHajeir Thanks for the failure info. I see a few things that I'll investigate. One is that the stack trace looks strange: I'm not sure why the interrupt handler for FastLED is being invoked while the other code is running. Another problem is the fact that rmt_tx_stop is not in IRAM -- we can't control where this function goes, but I can replace it with inline code.

HamzaHajeir, 2020-11-29:
I see, It's part of sdk and needs a workaround.

samguyer, 2020-11-30:
@HamzaHajeir and @zekageri 
I'm still curious about why the FastLED interrupt handler is getting invoked *while* the flash access code is running. That seems weird.

zekageri, 2020-11-30:
I'm curious too. I will investigate it further soon

samguyer, 2020-12-03:
@zekageri Does this crash happen fairly consistently? If I give you a patch for the FastLED library, can you try it out?

HamzaHajeir, 2020-12-03:
For my end, It's not fairly consistent. But it happens from time to time. 

If it would help to flash the device and and reboot it from time to time and watch the result over days. I'm happy to test out.

zekageri, 2020-12-03:
Of course. This happens all the time if i open a webpage hosted on the esp by littlefs. I don't tested with other fss.
I will be glad to try it out!

samguyer, 2020-12-04:
@zekageri and @HamzaHajeir I will push a change to my branch (https://github.com/samguyer/FastLED) that *might* fix the problem. I'd love to hear if it works. Thanks!

samguyer, 2020-12-04:
@zekageri and @HamzaHajeir  OK, get my latest version and see if that helps. All I did was inline the rmt functions that seem to be triggering the caching problem, that way all the code is in IRAM.

zekageri, 2020-12-04:
Thank you! I will try it out soon!

zekageri, 2020-12-04:
It seems to me this solved the problem. 
Some bail happening while the webpage is loading but nothing much. 
My ESP is restarting with Brownout detector when i open the serial monitor even with a plus capacitor on the power, but that's an other story. :D

I will report if something goes crazy but for now i close this isssue and thanks for your help. 
This lib is awesome and i really appreciate your work!

closed 2020-12-04 by zekageri

HamzaHajeir, 2020-12-06:
It seems to me that the issue is solved, too.
I've not encountered any reset on boot as before over these days. As I've tried many button resets, Power cycle and ESP.reset() ..

I think it's solved now

zekageri, 2020-12-07:
I did notice a strange behaviour. My color order is gone. 
Red Blue and Green is green with a slitghly different brightness.
Can you confirm this? Or is it just my leds?

HamzaHajeir, 2020-12-07:
Well, until this point all I've done is software integration of FastLED to
the codebase.

I'll feedback Once I connect the hardware for a real test.