FastLED/FastLED #1116: WS2812B issue with teensy 4.0

max246, 2020-11-21:
I have tried to run WS2812B dot leds with teensy and fastled but failed each time ( https://www.aliexpress.com/item/1897830725.html )

I have used the parallel system which  I set as follow:
#define DATA_PIN 19
FastLED.addLeds<1,WS2812B, DATA_PIN, GRB>(leds, NUM_LEDS);  // GRB ordering is typical

But kept flashing first led white and the second one yellow, I did try different combination, strip to 6 then 2, different teensy ( brand new one ) but kept not doing what it is supposed to do.

I have used the same code on an Arduino nano and works perfectly. 

i have added a 74HC246 as some people suggested because teensy 4.0 is running at 3.3v but still no luck.

marmilicious, 2020-11-21:
Hello @max246  We use github "issues" here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED

Put your test code on pastebin.com and share a link in your post. 
Also, please provide a link to the level shifter you are using if possible.

lpaolini, 2020-11-28:
With Teensy, together with FastLED, use [WS2812Serial](https://github.com/PaulStoffregen/WS2812Serial), which is non-blocking.

#include <WS2812Serial.h>
#define USE_WS2812SERIAL
#include <FastLED.h>

Then, when calling FastLED.addLeds use **WS2812SERIAL** instead of WS2812B.
Please note you can only use serial pins. (TX1, TX2, ...)

yajo10, 2020-11-30:
Hey, 
just for completeness I wanted to add that I had a similar Issue using teensy 4.1.
I had the snippet working quite well before on an esp32.

I had constantly withe color even if they should be off. 
If I wanted to make them red, they got blue-greenish. 

Switching to Serial worked like a charm and solved the Issue. Thanks for this thread!