FastLED/FastLED #1118: ESP32 (ld): undefined reference to `ESP32RMTController::ESP32RMTController

elliottkember, 2020-11-24:
Updated FastLED to get the new updates! Very excited about the new ESP32 functionality as that's the chipset I use.

I made a change to the `component.mk` as explained in https://github.com/FastLED/FastLED/pull/1108#issuecomment-732538248 - including the src directory. However, I'm now hitting this problem in the linking step:

```
Generating esp32.project.ld
LD /Users/elliott/Sites/soulmate/soulmate-builder-server/output/sketch/build/soulmate.elf
/Users/elliott/Sites/soulmate/soulmate-builder-server/output/sketch/build/main/libmain.a(main.o):(.literal._ZN15SoulmateLibrary5setupEv[SoulmateLibrary::setup()]+0x5c): undefined reference to `ESP32RMTController::ESP32RMTController(int, int, int, int)'
/Users/elliott/Sites/soulmate/soulmate-builder-server/output/sketch/build/main/libmain.a(main.o):(.literal._ZN19ClocklessControllerILi32ELi60ELi150ELi90EL6EOrder66ELi0ELb0ELi5EE13loadPixelDataER15PixelControllerILS0_66ELi1ELj4294967295EE[ClocklessController<32, 60, 150, 90, (EOrder)66, 0, false, 5>::loadPixelData(PixelController<(EOrder)66, 1, 4294967295u>&)]+0x0): undefined reference to `ESP32RMTController::getPixelBuffer(int)'
/Users/elliott/Sites/soulmate/soulmate-builder-server/output/sketch/build/main/libmain.a(main.o):(.literal._ZN19ClocklessControllerILi32ELi60ELi150ELi90EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELj4294967295EE[ClocklessController<32, 60, 150, 90, (EOrder)66, 0, false, 5>::showPixels(PixelController<(EOrder)66, 1, 4294967295u>&)]+0x0): undefined reference to `ESP32RMTController::showPixels()'
/Users/elliott/Sites/soulmate/soulmate-builder-server/output/sketch/build/main/libmain.a(main.o): In function `CLEDController::CLEDController()':
/Users/elliott/Sites/soulmate/soulmate-builder-server/soulmate-core/./SoulmateMatrix.h:130: undefined reference to `ESP32RMTController::ESP32RMTController(int, int, int, int)'
/Users/elliott/Sites/soulmate/soulmate-builder-server/output/sketch/build/main/libmain.a(main.o): In function `ClocklessController<32, 60, 150, 90, (EOrder)66, 0, false, 5>::loadPixelData(PixelController<(EOrder)66, 1, 4294967295u>&)':
/Users/elliott/Sites/soulmate/soulmate-builder-server/soulmate-core/./SoulmateMatrix.h:130: undefined reference to `ESP32RMTController::getPixelBuffer(int)'
/Users/elliott/Sites/soulmate/soulmate-builder-server/output/sketch/build/main/libmain.a(main.o): In function `ClocklessController<32, 60, 150, 90, (EOrder)66, 0, false, 5>::showPixels(PixelController<(EOrder)66, 1, 4294967295u>&)':
/Users/elliott/Sites/soulmate/soulmate-builder-server/soulmate-core/./SoulmateMatrix.h:130: undefined reference to `ESP32RMTController::showPixels()'
collect2: error: ld returned 1 exit status

```

What's weird is that it compiles okay, but it's the linker (ld) that has the problem. It seems to be triggered when addLeds is called as follows:

`FastLED.addLeds<WS2812B, 32, RGB>(leds, N_LEDS);`

elliottkember, 2020-11-24:
PR 1119 will hopefully take care of this by adding the following lines to the `component.mk`:

```
COMPONENT_ADD_INCLUDEDIRS := ./src src/platforms/esp/32
COMPONENT_SRCDIRS := ./src src/platforms/esp/32
```

samguyer, 2020-11-25:
@elliottkember Thank you! I don't do any work in PlatformIO, so I don't notice right away if there's a problem. I guess I should start...

elliottkember, 2020-11-25:
@samguyer Neither do I, though I suspect it's the way to go. I just use my Soulmate IDE.

closed 2020-12-30 by elliottkember

Un3qual, 2021-01-26:
I am on version 3.4.0 using PlatformIO and I still get this error. I haven't had any problems when using FastLED with just the Arduino library, but when using it with esp-idf I had to fix a bunch of stuff and now I'm stuck with this error. 

```
/Users/admin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/az-delivery-devkit-v4/esp-idf/src/Blink.cpp.o:(.literal._ZN19ClocklessControllerILi32ELi40ELi100ELi60EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELj4294967295EE[ClocklessController<32, 40, 100, 60, (EOrder)66, 0, false, 5>::showPixels(PixelController<(EOrder)66, 1, 4294967295u>&)]+0x0): undefined reference to `ESP32RMTController::getPixelBuffer(int)'
/Users/admin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/az-delivery-devkit-v4/esp-idf/src/Blink.cpp.o:(.literal._ZN19ClocklessControllerILi32ELi40ELi100ELi60EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELj4294967295EE[ClocklessController<32, 40, 100, 60, (EOrder)66, 0, false, 5>::showPixels(PixelController<(EOrder)66, 1, 4294967295u>&)]+0x4): undefined reference to `ESP32RMTController::showPixels()'
/Users/admin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/az-delivery-devkit-v4/esp-idf/src/Blink.cpp.o:(.literal._Z5setupv+0x5c): undefined reference to `ESP32RMTController::ESP32RMTController(int, int, int, int)'
/Users/admin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/az-delivery-devkit-v4/esp-idf/src/Blink.cpp.o: in function `ClocklessController<32, 40, 100, 60, (EOrder)66, 0, false, 5>::showPixels(PixelController<(EOrder)66, 1, 4294967295u>&)':
/Users/admin/Documents/PlatformIO/Projects/210125-135537-espidf-arduino-blink/.pio/libdeps/az-delivery-devkit-v4/FastLED/src/platforms/esp/32/clockless_rmt_esp32.h:332: undefined reference to `ESP32RMTController::getPixelBuffer(int)'
/Users/admin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/admin/Documents/PlatformIO/Projects/210125-135537-espidf-arduino-blink/.pio/libdeps/az-delivery-devkit-v4/FastLED/src/platforms/esp/32/clockless_rmt_esp32.h:382: undefined reference to `ESP32RMTController::showPixels()'
/Users/admin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/az-delivery-devkit-v4/esp-idf/src/Blink.cpp.o: in function `setup()':
/Users/admin/Documents/PlatformIO/Projects/210125-135537-espidf-arduino-blink/.pio/libdeps/az-delivery-devkit-v4/FastLED/src/controller.h:69: undefined reference to `ESP32RMTController::ESP32RMTController(int, int, int, int)'
collect2: error: ld returned 1 exit status
*** [.pio/build/az-delivery-devkit-v4/firmware.elf] Error 1
```

Un3qual, 2021-01-26:
Putting
```c
#ifndef ESP32
#define ESP32
#endif
```
at the top of `src/platforms/esp/32/clockless_rmt_esp32.cpp` fixed it. Where does the name of the platform usually get defined?

thomasbeauvais, 2021-07-12:
Using IDF this doesn't really work. I have the same errors.

`[ 99%] Linking CXX executable led-poc.elf
/xxx/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(main.cpp.obj):(.literal._ZN19ClocklessControllerILi0ELi52ELi52ELi103EL6EOrder10ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_10ELi1ELj4294967295EE[ClocklessController<0, 52, 52, 103, (EOrder)10, 0, false, 5>::showPixels(PixelController<(EOrder)10, 1, 4294967295u>&)]+0x0): undefined reference to 'ESP32RMTController::getPixelBuffer(int)'
/xxx/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(main.cpp.obj):(.literal._ZN19ClocklessControllerILi0ELi52ELi52ELi103EL6EOrder10ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_10ELi1ELj4294967295EE[ClocklessController<0, 52, 52, 103, (EOrder)10, 0, false, 5>::showPixels(PixelController<(EOrder)10, 1, 4294967295u>&)]+0x4): undefined reference to 'ESP32RMTController::showPixels()'
/xxx/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(main.cpp.obj):(.literal._Z5setupv+0x20): undefined reference to 'ESP32RMTController::ESP32RMTController(int, int, int, int, int, int)'
/xxx/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(main.cpp.obj): in function 'ClocklessController<0, 52, 52, 103, (EOrder)10, 0, false, 5>::loadPixelData(PixelController<(EOrder)10, 1, 4294967295u>&)':
/xxx/led/main/platforms/esp/32/clockless_rmt_esp32.h:345: undefined reference to 'ESP32RMTController::getPixelBuffer(int)'
/xxx/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(main.cpp.obj): in function 'ClocklessController<0, 52, 52, 103, (EOrder)10, 0, false, 5>::showPixels(PixelController<(EOrder)10, 1, 4294967295u>&)':
/xxx/led/main/platforms/esp/32/clockless_rmt_esp32.h:367: undefined reference to 'ESP32RMTController::showPixels()'
/xxx/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(main.cpp.obj): in function 'ClocklessController<0, 52, 52, 103, (EOrder)10, 0, false, 5>::ClocklessController()':
/xxx/led/main/platforms/esp/32/clockless_rmt_esp32.h:325: undefined reference to 'ESP32RMTController::ESP32RMTController(int, int, int, int, int, int)`

tomnz, 2022-04-21:
I'm also encountering this. Strangely, this only seems to occur when trying to use a three-pin chip like `NEOPIXEL` or `WS2811`. Code links just fine with `DOTSTAR` or `SK9822`. Workarounds above such as hacking in a `#define ESP32` didn't seem to work. Is three-wire supported for ESP32?

tomnz, 2022-04-21:
Meant to add - I am also on esp-idf with a somewhat custom CMake structure. I'll have to try playing around with the component definition... I don't think it's a coincidence that the clockless code is the only stuff with a source (cpp) file in the esp32 folder.

samguyer, 2022-04-21:
@tomnz The symbol you're missing is in src/platforms/esp/32/clockless_rmt_esp32.cpp. This file contains the driver for the clockless LEDs (the three-pin LEDs) for ESP32. An important difference between this driver and the others is that it is in a cpp file, not a header file, so it is not automatically included just by putting #include in your sketch. It needs to be compiled separately and linked in. You'll see the special case handling in library.json and component.mk files. I think you just need to make sure you include these extra files (src/platforms/esp/32/*.cpp) in your CMake config.

tomnz, 2022-04-21:
Ahh yes, this was exactly the issue, thanks! To describe my situation and solution in case this is helpful for others:

* Due to the way esp-idf's [build system](https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-guides/build-system-cmake.html) works, I am not directly including/using FastLED's `component.mk`.
* Instead, the entire FastLED source tree is embedded under `components/fastled/FastLED` (i.e. a subfolder) and in the parent folder `components/fastled` I have a custom `CMakeLists.txt` and `component.mk`. I've pasted the updated versions of these below.
* These custom configurations neglected to denote `FastLED/src/platforms/esp/32` as a source directory, and therefore when switching to the clockless chipset the linker fails to find the implementations in `clockless_rmt_esp32.cpp`. This was not a problem with a clocked chip, as that was entirely contained in header files.

My updated files are as follows:

`CMakeLists.txt`
```
set(COMPONENT_SRCS "FastLED/src/FastLED.cpp" "FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp")
set(COMPONENT_ADD_INCLUDEDIRS "FastLED/src" "FastLED/src/platforms/esp/32")

register_component()
```

`component.mk`
```
COMPONENT_ADD_INCLUDEDIRS := FastLED/src FastLED/src/platforms/esp/32
COMPONENT_SRCDIRS := FastLED/src FastLED/src/platforms/esp/32
```

I'm actually not certain of the necessity of `CMakeLists.txt` in my build configuration. If it turns out not to be needed, then FastLED already defines a `component.mk` and the subfolder / custom build files are unnecessary.

Code is now compiling and linking flawlessly!