FastLED/FastLED #1123: Unused variable warning in: clockless_rmt_esp32.h on compile

DenverClarke, 2020-11-27:
Hey there. I don't know if this is a problem or not, but I'm getting:
{
	"resource": "/c:/Users/Syx/Documents/PlatformIO/Projects/Starshroud ESP/.pio/libdeps/heltec_wifi_kit_32/FastLED/platforms/esp/32/clockless_rmt_esp32.h",
	"owner": "cpp",
	"severity": 4,
	"message": "unused variable 't1' [-Wunused-variable]",
	"startLineNumber": 511,
	"startColumn": 22,
	"endLineNumber": 511,
	"endColumn": 22
}

Probably not important, just thought I would make a post in case it is important.

patfelst, 2020-12-12:
I'm getting this too, on an ESP32 in PlatformIO.

patfelst, 2020-12-12:
You can just comment out this line
https://github.com/FastLED/FastLED/blob/fefc5f563a33963b9998f0475ad050813b30adcb/platforms/esp/32/clockless_rmt_esp32.h#L511

Where the t1 variable is declared and the warning goes away. This will obviously come back next time you update the FastLED library.

DenverClarke, 2020-12-12:
Thanks for the reply! It doesn't seem to be making an any trouble, but I wonder what it does, since I'm compiling for clockless LEDs.

patfelst, 2020-12-12:
Same here, i.e. the code works, but I don't like seeing any errors in the compiler output, as then I might miss an important one in my own code! I'm investigating tuning the build options to suppress library warnings.

samguyer, 2020-12-14:
@DenverClarke Thanks for the issue report. I believe we fixed this issue in the latest version of the code, but please let me know if you continue to have a problem.

closed 2020-12-14 by samguyer

DenverClarke, 2020-12-14:
Sweet. Thanks for your hard work on this project. I really appreciate it.