FastLED/FastLED #1125: FastLED can't be built anylonger under PlatformIO

sixtyfive, 2020-11-29:
There's dozens of these:

```
FastLED/src/platforms/avr/clockless_trinket.h:334:24: error: expected primary-expression before '__attribute__'
```

5chmidti, 2020-11-29:
@samguyer you basically reverted #961 in 26e222e. I guess that was by accident, but at least, that is where this comes from.

sixtyfive, 2020-11-29:
In case it helps:

```bash
~/.platformio/packages/toolchain-atmelavr/bin/avr-gcc --version
avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.2_1759) 5.4.0
```

5chmidti, 2020-11-29:
I'm sorry, I think it's actually from #1050. That got merged after Oct 26.

kriegsman, 2020-11-30:
Thanks for reporting this; we're looking at it now.

kriegsman, 2020-12-01:
OK, I think that https://github.com/FastLED/FastLED/commit/dd3723664847e2aa3c936fc03fb038f9e5f985e8 may fix this; it's been merged in.  I don't have PlatformIO set up on the machine I have here right now, but I suspect this will get past that problem.  We had some earlier merge problems which we're working through, but I think this should help.

sixtyfive, 2020-12-02:
Hi @kriegsman, thank you, it's building fine again now! :-)

By the way, what's with these? Are they on purpose?

```cpp
In file included from include/rgbled.h:4:0,
                 from src/bmp388.cpp:7:
lib/FastLED/src/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.003
 #    pragma message "FastLED version 3.003.003"
```

closed 2020-12-02 by sixtyfive