FastLED/FastLED #1126: FastLED.Show() has a negative impact on digitalRead(GPIO_NUM_0)

alf45tar, 2020-11-30:
Using the latest `master` branch after a FastLED.show() call the next calls to digitalRead(0) return always LOW.
The following code prints the hash sign indefinitively even if the BOOT button (GPIO0) is not pressed.

````
#include <Arduino.h>
#include "FastLED.h"

CRGB fastleds[20];

void setup()
{
  Serial.begin(115200);

  FastLED.addLeds<NEOPIXEL, 5>(fastleds, 20);
  for (byte l = 0; l < 20; l++)
    fastleds[l] = CRGB::Red;
  FastLED.show();

  pinMode(0, INPUT_PULLUP);
}

void loop()
{
  if (digitalRead(0) == LOW) {
      Serial.print("#");
    }
}
````

The above code works as expected (print hash sign only when BOOT button is pressed) in version 3.3.3.

Jueff, 2020-12-07:
if it's on an ESP32 this workaround may help 
https://github.com/FastLED/FastLED/issues/1130#issuecomment-737310821

samguyer, 2020-12-30:
@alf45tar This problem should be fixed in the latest release. Let me know if you're still having problems.

alf45tar, 2020-12-30:
I confirm it is ok now.
Thanks

closed 2020-12-30 by alf45tar