FastLED/FastLED #1130: ESP32 GPIO get unuseable

Jueff, 2020-12-02:
I discovered an issue in the current trunk, that GPIO0 pin of ESP32 can't be used as output anymore, after calling FastLed.show the first time.

I found out that it must have something to do with
void ESP32RMTController::init()
...
        rmt_tx.gpio_num = gpio_num_t(0);  // The particular pin will be assigned later
....
If I change to e.g.  gpio_num_t(22) GPIO 0 can be used as output again.
Seems the pin is set to another kind of operation, but isn't reset to default  when assigning the correct pin.

Jueff, 2020-12-02:
As a workaround I added 

`gpio_matrix_out(gpio_num_t(0), 0x100, false, false);`

in my code after the first FastLED.show call.

samguyer, 2020-12-15:
@Jueff Thanks for the issue report! I should be able to fix that problem in a more permanent way.

samguyer, 2020-12-15:
@Jueff I just pushed a fix for this issue, so I'm closing it. Thanks!

closed 2020-12-15 by samguyer

Jueff, 2020-12-15:
Thanks for the quick fix!