FastLED/FastLED #1131: Colors are suddenly changed.

zekageri, 2020-12-06:
**I'm on esp32. i downloaded the latest code from github and uploaded the code with it. I changed nothing on the setup and the colors are changed somehow.** 

### Here is my setup:

```
#define LED_PIN     27
#define NUM_LEDS    90
#define CHIPSET     WS2812
#define COLOR_ORDER GRB //GBR //RBG //RGB //BRG //BGR // I have tried every possibilities.
CRGB leds[NUM_LEDS];

void setup(){
  pinMode(LED_PIN, OUTPUT);
  FastLED.addLeds<CHIPSET, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalSMD5050 );
  FastLED.setBrightness( BRIGHTNESS );
}
```

I'm using ws2812B with 90 leds. A big power supply ( capable of 10A ). The led is 5v. GND and esp GND are wired together. ESP get's power from usb, led gets power from big power supply. 
Yellow became green and etc. Used GRB color order last.

samguyer, 2020-12-07:
@zekageri Are you using my repo (under samguyer) or the one here under FastLED?

Can you describe exactly what the color changes are? For example, if you just give it all red, then all green, then all blue, what do you see?

zekageri, 2020-12-07:
I'am using the RMT controller crash fix repo, where you put the instructions to IRAM. 

The colors are:
Green is a beautiful green

Yellow is green too with slightly lower brightness

Red is green again with a really low brightness

samguyer, 2020-12-07:
@zekageri That is strange. Let me test it on my end.

zekageri, 2020-12-07:
> @zekageri Are you using my repo (under samguyer) or the one here under FastLED?
> 
> Can you describe exactly what the color changes are? For example, if you just give it all red, then all green, then all blue, what do you see?

Wich one should i use?

zekageri, 2020-12-07:
Using your repo ( under your name ) the RMT controller problem is gone but almost every color is green. Under this repo, the RMT controller restart is still there.

samguyer, 2020-12-07:
With exactly the same hardware, just different versions of FastLED?

zekageri, 2020-12-07:
Yes.

zekageri, 2020-12-07:
Everything is the same except the fatLED lib. One is without RMT ( IRAM ) fix one is with it

samguyer, 2020-12-07:
@zekageri I'll take a look today. That should not happen!

zekageri, 2020-12-07:
I noticed that if i go for full brightness on a html default colorpicker i get greens. But if i slightly lower the "value" the colors are okay.

samguyer, 2020-12-07:
I think I figured it out. I need to back out one of the changes I made.

samguyer, 2020-12-07:
OK, I just pushed to my repo. See if that works now. I will look more carefully at whether there is a longer-term change we can make.

zekageri, 2020-12-07:
Testing

zekageri, 2020-12-07:
**Integer Divided by zero with new lib on your repo. Last is working same code same hardware**.

```
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e5409  PS      : 0x00060f30  A0      : 0x800e57be  A1      : 0x3ffb1e30
A2      : 0x00000133  A3      : 0x00000000  A4      : 0x00000032  A5      : 0x3ffbec78
A6      : 0x00004e20  A7      : 0x60013000  A8      : 0x00000113  A9      : 0x3ffb1e10
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffb854c  A13     : 0x3ffc278c
A14     : 0x00002710  A15     : 0x3ffbec78  SAR     : 0x0000001c  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x400e5409:0x3ffb1e30 0x400e57bb:0x3ffb1e60 0x400ddd0d:0x3ffb1e80 0x400ddd61:0x3ffb1ea0 0x400dde6d:0x3ffb1ec0 0x400de551:0x3ffb1ee0 0x400de605:0x3ffb1f00 0x400de689:0x3ffb1f20 0x400d3c9d:0x3ffb1f40 0x400d3d1b:0x3ffb1f90 0x400e6421:0x3ffb1fb0 0x4008969d:0x3ffb1fd0
  #0  0x400e5409:0x3ffb1e30 in i2cProcQueue at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-i2c.c:1279
  #1  0x400e57bb:0x3ffb1e60 in i2cWrite at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-i2c.c:1621
  #2  0x400ddd0d:0x3ffb1e80 in TwoWire::writeTransmission(unsigned short, unsigned char*, unsigned short, bool) at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src/Wire.cpp:302
  #3  0x400ddd61:0x3ffb1ea0 in TwoWire::endTransmission(bool) at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src/Wire.cpp:302
  #4  0x400dde6d:0x3ffb1ec0 in TwoWire::endTransmission() at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src/Wire.cpp:302
  #5  0x400de551:0x3ffb1ee0 in Adafruit_ADXL345_Unified::read16(unsigned char) at lib\Adafruit_ADXL345-master/Adafruit_ADXL345_U.cpp:130
  #6  0x400de605:0x3ffb1f00 in Adafruit_ADXL345_Unified::getZ() at lib\Adafruit_ADXL345-master/Adafruit_ADXL345_U.cpp:182
  #7  0x400de689:0x3ffb1f20 in Adafruit_ADXL345_Unified::getEvent(sensors_event_t*) at lib\Adafruit_ADXL345-master/Adafruit_ADXL345_U.cpp:330
  #8  0x400d3c9d:0x3ffb1f40 in adxl_Loop() at lib/FastLED-master/src/pixeltypes.h:459
  #9  0x400d3d1b:0x3ffb1f90 in loop() at lib/FastLED-master/src/pixeltypes.h:459
  #10 0x400e6421:0x3ffb1fb0 in loopTask(void*) at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:19
  #11 0x4008969d:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

zekageri, 2020-12-07:
I do have two active i2c communication. But that serial crash split is wrong in so many levels. With last fastled lib the same code ( with i2c communication ) is working properly. I have an RTC and an ADXL module on there.

samguyer, 2020-12-07:
@zekageri I'd be surprised if that were actually related. I'm not sure how to diagnose it without any FastLED stuff in the call stack.

zekageri, 2020-12-07:
I comment out the i2c stuff and we will see if there is any useful serial information

zekageri, 2020-12-07:
Lol no crash without the ADXL codes. Interesting.

zekageri, 2020-12-07:
Okay, i put back the i2c stuff and the error gone. Colors are good, everything is working. I will investigate it further.

samguyer, 2020-12-07:
@zekageri Great!

zekageri, 2020-12-07:
**I don't know if the Async Lib is that messy or something goes bad with the fastLED still., :(**

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x400e36bc  PS      : 0x00060034  A0      : 0x8008119c  A1      : 0x3ffbe790  
A2      : 0x3ffc1330  A3      : 0x3ffc2760  A4      : 0xc9b28afa  A5      : 0x3ffd96a0  
A6      : 0x00000020  A7      : 0xffffffff  A8      : 0x80080ff9  A9      : 0x00000001  
A10     : 0x3ffc2760  A11     : 0x00008e8d  A12     : 0x0000000a  A13     : 0x00060820  
A14     : 0x00000003  A15     : 0x00060823  SAR     : 0x00000018  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x400876cd  EPC2    : 0x00000000  EPC3    : 0x400937e2  EPC4    : 0x400e36bc  

Backtrace: 0x400e36bc:0x3ffbe790 0x40081199:0x3ffbe7b0 0x400853e1:0x3ffbe7d0 0x400937df:0x3ffd96e0 0x40087d04:0x3ffd9700 0x4013e9b3:0x3ffd9770 0x400e31d0:0x3ffd97a0 0x400dfe15:0x3ffd97d0 0x400e1d19:0x3ffd9800 0x400df473:0x3ffd9840 0x4010ffca:0x3ffd9870 0x4000bdbb:0x3ffd9890 0x40001125:0x3ffd98b0 0x400594e9:0x3ffd98d0 0x40153602:0x3ffd98f0 0x40153675:0x3ffd9920 0x40146223:0x3ffd9940 0x40153fef:0x3ffd9960 0x400da625:0x3ffd9980 0x400db7a9:0x3ffd99a0 0x400dbc3a:0x3ffd9a30 0x400d7c5a:0x3ffd9a80 0x400d7c65:0x3ffd9aa0 0x401457e6:0x3ffd9ac0 0x40145acb:0x3ffd9af0 0x40145add:0x3ffd9b20 0x40145c1b:0x3ffd9b40 0x4008969d:0x3ffd9b70
  #0  0x400e36bc:0x3ffbe790 in Print::print(unsigned int, int) at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\cores\esp32/Print.cpp:234
  #1  0x40081199:0x3ffbe7b0 in ESP32RMTController::interruptHandler(void*) at lib\FastLED-master\src\platforms\esp\32/clockless_rmt_esp32.cpp:420
  #2  0x400853e1:0x3ffbe7d0 in _xt_medint3 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1323
  #3  0x400937df:0x3ffd96e0 in esp_rom_spiflash_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:546
  #4  0x40087d04:0x3ffd9700 in spi_flash_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:567
  #5  0x4013e9b3:0x3ffd9770 in esp_partition_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/partition.c:242
  #6  0x400e31d0:0x3ffd97a0 in littlefs_api_read at lib\LITTLEFS\src/littlefs_api.c:22  
  #7  0x400dfe15:0x3ffd97d0 in lfs_bd_read at lib\LITTLEFS\src/lfs.c:3459
  #8  0x400e1d19:0x3ffd9800 in lfs_file_read at lib\LITTLEFS\src/lfs.c:3459
  #9  0x400df473:0x3ffd9840 in vfs_littlefs_read at lib\LITTLEFS\src/esp_littlefs.c:231 
  #10 0x4010ffca:0x3ffd9870 in esp_vfs_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:972
  #11 0x4000bdbb:0x3ffd9890 in ?? ??:0
  #12 0x40001125:0x3ffd98b0 in ?? ??:0
  #13 0x400594e9:0x3ffd98d0 in ?? ??:0
  #14 0x40153602:0x3ffd98f0 in _fread_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fread.c:262
  #15 0x40153675:0x3ffd9920 in fread at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fread.c:301  #16 0x40146223:0x3ffd9940 in VFSFileImpl::read(unsigned char*, unsigned int) at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:127 (discriminator 1)
  #17 0x40153fef:0x3ffd9960 in fs::File::read(unsigned char*, unsigned int) at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\libraries\FS\src/FS.cpp:269 
  #18 0x400da625:0x3ffd9980 in AsyncFileResponse::_fillBuffer(unsigned char*, unsigned int) at lib\ESPAsyncWebServer\src/WebResponses.cpp:699
  #19 0x400db7a9:0x3ffd99a0 in AsyncAbstractResponse::_fillBufferAndProcessTemplates(unsigned char*, unsigned int) at lib\ESPAsyncWebServer\src/WebResponses.cpp:699
  #20 0x400dbc3a:0x3ffd9a30 in AsyncAbstractResponse::_ack(AsyncWebServerRequest*, unsigned int, unsigned int) at lib\ESPAsyncWebServer\src/WebResponses.cpp:699
  #21 0x400d7c5a:0x3ffd9a80 in AsyncWebServerRequest::_onPoll() at lib\ESPAsyncWebServer\src/WebRequest.cpp:797
  #22 0x400d7c65:0x3ffd9aa0 in std::_Function_handler<void (void*, AsyncClient*), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*)#9}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&) at lib\ESPAsyncWebServer\src/WebRequest.cpp:797
      (inlined by) _M_invoke at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #23 0x401457e6:0x3ffd9ac0 in std::function<void (void*, AsyncClient*)>::operator()(void*, AsyncClient*) const at lib\AsyncTCP\src/AsyncTCP.cpp:1079
  #24 0x40145acb:0x3ffd9af0 in AsyncClient::_poll(tcp_pcb*) at lib\AsyncTCP\src/AsyncTCP.cpp:1079
  #25 0x40145add:0x3ffd9b20 in AsyncClient::_s_poll(void*, tcp_pcb*) at lib\AsyncTCP\src/AsyncTCP.cpp:1079
  #26 0x40145c1b:0x3ffd9b40 in _async_service_task(void*) at lib\AsyncTCP\src/AsyncTCP.cpp:1079
      (inlined by) _async_service_task at lib\AsyncTCP\src/AsyncTCP.cpp:197
  #27 0x4008969d:0x3ffd9b70 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57
```

zekageri, 2020-12-07:
How can one program produce so much different errors in one day? xd

samguyer, 2020-12-07:
@zekageri I hear you! I think that one is also my fault. The print function is probably not in IRAM (and I really shouldn't be calling it from within an ISR). I'll fix and push again!

But that you for all of your detailed feedback -- it is super-helpful to have users who are willing to try things out and report back.

samguyer, 2020-12-08:
@zekageri OK, try pulling my code one more time.

zekageri, 2020-12-08:
Thanks I will try it as soon as I can!

zekageri, 2020-12-09:
Seems stable for now. 
I will test it more today.

samguyer, 2020-12-16:
@zekageri Can I close this issue as fixed?

zekageri, 2020-12-16:
Yes. Sorry. Everything is fine. Thanks for everything!

closed 2020-12-16 by zekageri