FastLED/FastLED #1132: ColorFromPalette not returns first index (0) colour as max index (255)

tmargot, 2020-12-06:
Hi,

I am new to FastLED but I have defined a simple palette from red (FF0000) to green (00FF00), but noticed that calling ColorFromPalette with palette index of 0 or 255 produces the same color?

Arduino: Nano
LED: WS2812b 

My simplified code:

`#include <FastLED.h>
#define DATA_PIN 6
#define NUM_LEDS 2
DEFINE_GRADIENT_PALETTE( christmas_gp ) {
  0,      255,  0,    0,   //red
  255,    0,    255,  0,   //green
}; 
CRGBPalette16 palChristmas = christmas_gp;

CRGB leds[NUM_LEDS];

void setup() {   
  delay(1000);                                                // Soft startup to ease the flow of electrons. 
  FastLED.clear();    
  FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, NUM_LEDS);  // GRB ordering is typical
  leds[0] = ColorFromPalette( palChristmas, 255, 255, LINEARBLEND); // shows red
  leds[1] = ColorFromPalette( palChristmas, 0, 255, LINEARBLEND); // also show red?????
  FastLED.show();  
}    

void loop() { 
}
`

Any help would be much appreciated.

marmilicious, 2020-12-06:
@tmargot  See the comment here.  Is that what you're seeing?  

https://github.com/FastLED/FastLED/issues/515

tmargot, 2020-12-06:
@marcmerlin  - Thanks so much for your answer. This is exactly what I am seeing, I will have to check is passing palette location as 240 will resolve the issue, but I would have assume passing blending as NOBLEND would have resolved the issue, it didn't,  but I may be misunderstanding.

marcmerlin, 2020-12-06:
probably meant for @marmilicious  who is another fine Marc :)

tmargot, 2020-12-06:
Oh yes! Sorry!

tmargot, 2020-12-06:
I will follow the advice given in the link, thank you

marmilicious, 2020-12-06:
I must not have hit send earlier...  
@tmargot 
If you add another anchor print to your gradient at 240 then you can go from 0 to 240 and it will be pure green at that index.  

DEFINE_GRADIENT_PALETTE( christmas_gp ) {
0, 255, 0, 0, //red
240, 0, 255, 0,
255, 0, 255, 0, //green
};

If you have more general questions please come ask on the FastLED Reddit group.  Feel free to close this ticket if you're good here.