FastLED/FastLED #1133: Error compiling for board Digispark (Default -16.5mhz)

gokalpsterr, 2020-12-07:
I am using a Windows10 pc and trying to program a Digispark Attiny85 using the Arduino 1.8.5 IDE(with FastLED version 3.3.3).
In my project, I need to use an IMU(gy521 using I2C) and 16 LED NeoPixel ring that I want to use the FastLED library for.

I am sorry if this has been already asked but as far as I checked on this Github page and many other Arduino forums, I could not find a solution.

Here is the error that I am getting:

````
Arduino: 1.8.5 (Windows 10), Board: "Digispark (Default - 16.5mhz)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\gokal\AppData\Local\Arduino15\packages -hardware C:\Users\gokal\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\gokal\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\gokal\Documents\Arduino\libraries -fqbn=digistump:avr:digispark-tiny -ide-version=10805 -build-path C:\Users\gokal\AppData\Local\Temp\arduino_build_954355 -warnings=none -build-cache C:\Users\gokal\AppData\Local\Temp\arduino_cache_545266 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.micronucleus.path=C:\Users\gokal\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4 -prefs=runtime.tools.avr-gcc.path=C:\Users\gokal\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -verbose C:\Users\gokal\AppData\Local\Temp\arduino_modified_sketch_805229\sketch_dec06a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\gokal\AppData\Local\Arduino15\packages -hardware C:\Users\gokal\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\gokal\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\gokal\Documents\Arduino\libraries -fqbn=digistump:avr:digispark-tiny -ide-version=10805 -build-path C:\Users\gokal\AppData\Local\Temp\arduino_build_954355 -warnings=none -build-cache C:\Users\gokal\AppData\Local\Temp\arduino_cache_545266 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.micronucleus.path=C:\Users\gokal\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4 -prefs=runtime.tools.avr-gcc.path=C:\Users\gokal\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -verbose C:\Users\gokal\AppData\Local\Temp\arduino_modified_sketch_805229\sketch_dec06a.ino
Using board 'digispark-tiny' from platform in folder: C:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7
Using core 'tiny' from platform in folder: C:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7
Detecting libraries used...
"C:\Users\gokal\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10805 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR  "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny" "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\variants\digispark" "C:\Users\gokal\AppData\Local\Temp\arduino_build_954355\sketch\sketch_dec06a.ino.cpp" -o "nul"
"C:\Users\gokal\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10805 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR  "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny" "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\variants\digispark" "-IC:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src" "C:\Users\gokal\AppData\Local\Temp\arduino_build_954355\sketch\sketch_dec06a.ino.cpp" -o "nul"
"C:\Users\gokal\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10805 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR  "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny" "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\variants\digispark" "-IC:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src" "C:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src\FastLED.cpp" -o "nul"
"C:\Users\gokal\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10805 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR  "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny" "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\variants\digispark" "-IC:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src" "C:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src\bitswap.cpp" -o "nul"
"C:\Users\gokal\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10805 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR  "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny" "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\variants\digispark" "-IC:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src" "C:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src\colorpalettes.cpp" -o "nul"
"C:\Users\gokal\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10805 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR  "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny" "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\variants\digispark" "-IC:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src" "C:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src\colorutils.cpp" -o "nul"
"C:\Users\gokal\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10805 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR  "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny" "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\variants\digispark" "-IC:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src" "C:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src\hsv2rgb.cpp" -o "nul"
"C:\Users\gokal\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10805 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR  "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny" "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\variants\digispark" "-IC:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src" "C:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src\lib8tion.cpp" -o "nul"
"C:\Users\gokal\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10805 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR  "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny" "-IC:\Users\gokal\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\variants\digispark" "-IC:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src" "C:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src\noise.cpp" -o "nul"
[... cut ...]
C:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src/platforms/avr/clockless_trinket.h:286:46: error: impossible constraint in 'asm'

 #define DONE asm __volatile__("2:" ASM_VARS );

                                              ^

C:\Users\gokal\Documents\Arduino\libraries\FastLED-master\src/platforms/avr/clockless_trinket.h:460:4: note: in expansion of macro 'DONE'

    DONE;

    ^

Using library FastLED-master at version 3.3.3 in folder: C:\Users\gokal\Documents\Arduino\libraries\FastLED-master 
exit status 1
Error compiling for board Digispark (Default - 16.5mhz).
````

kriegsman, 2020-12-08:
Thanks for the report, we'll take a look at it.  I'm pretty sure that FastLED and the Digispark worked okay together before, so it's probably something minor.  Thanks again for reporting it; please add any comments if you make any progress on your end.

HaxNobody, 2020-12-14:
Reverting to FastLED 3.3.2 worked for me. Not sure what change causes the issue.