FastLED/FastLED #1134: STM32duino support?

Pararera, 2020-12-07:
Hi,
does this library supports STM32 chips and boards with STM32duino Core? If not, is there any plan to do it?
Thanks!

baniaque, 2021-01-09:
This should be main issue to resolve, there are many STM32 controllers that are great for using with LED, but most if them are able to use only with STM32duino, Clarks project developing to slow and not support many popular boards.

btply, 2021-01-30:
+1 for better STM32 Support

HappySamuel, 2023-04-21:
+1 for better STM32 support

zackees, 2026-06-03:
Triage update from fbuild side, tracked under FastLED/fbuild#385.

STM32duino support status from the fbuild backend (the new Rust build runner that ships with FastLED for `--platformio`-replacement builds):

| STM32 family | fbuild status (Arduino framework, ststm32 platform) |
|---|---|
| STM32F0xx (Cortex-M0) | **Gap** - tracked by FastLED/FastLED#750 |
| STM32F1xx (Cortex-M3) | Supported - smoke builds pass for bluepill_f103c8 / maple_mini_b20 / hy_tinystm103tb |
| STM32F4xx (Cortex-M4F) | Supported - smoke builds pass for blackpill_f411ce / nucleo_f429zi |
| STM32L4xx (Cortex-M4F, no DSP) | **Gap** - tracked by FastLED/FastLED#975 |
| STM32H7xx (Cortex-M7) | Supported - smoke builds pass for giga_r1_m7 (same MCU as Portenta H7 M7) |
| STM32U5xx (Cortex-M33F) | Supported (config in place) |

So yes - STM32duino-style builds are supported across the most common F1/F4/H7 die families today. Remaining FastLED-side platform header work is independent of fbuild.

See FastLED/fbuild#385 and PR FastLED/fbuild#396 for evidence and the gap-tracking fixtures.