FastLED/FastLED #1135: WS2812B + Arduino Uno + Adalight + Prismatik Screen Capture - flashing issue

PetrLukes, 2020-12-08:
Hello,

I have setup mentioned above, and after around 10 mins of usage, I have constant flashing same as on video below
https://www.youtube.com/watch?v=xHdZHHp_NBI&ab_channel=PetrLuke%C5%A1

I think it's not Prismatik related, since the flashing is happening also when the program is not running.
I'm using the latest version of FastLED with Adalight modification.

Any sort of help is grealty appreciated

Benik3, 2020-12-08:
I recommend to try to use NeoPixelBus library with Uart ("non blocking") method :)
https://github.com/Benik3/Adalight_WS2812_ESP8266_Non-blocking

PetrLukes, 2020-12-08:
Unfortunately, I cant make this version of NeoPixelBus work. Now I'm using FASTLED

sblantipodi, 2020-12-14:
@PetrLukes I would try the Luciferin suite.
It consists of a firmware and a PC software, it does not have those problems.

https://github.com/sblantipodi/firefly_luciferin

Benik3, 2020-12-14:
> 
> 
> Unfortunately, I cant make this version of NeoPixelBus work. Now I'm using FASTLED

Yeah I got that you are using Fastled, but I wanted to see the whole code...
I use Arduino micro with FastLed and APA102 + Prismatik for more then 3 years without problem.
Also what you didn't get to work on the NeoPixelBus? 

@sblantipodi I don't see firmware for UNO, does it support only ESP?