FastLED/FastLED #1136: FastLED.setBrightness glitch on Arduino.

johnty57, 2020-12-10:
HI,
I'm trying to compile the following code on an Arduino.

`#include <FastLED.h>

#define NUM_LEDS 8
#define LED_PIN 5

CRGB leds[NUM_LEDS];

void setup() {
  // put your s0etup code here, to run once:
Serial.begin(9600);

FastLED.addLeds<WS2812B, LED_PIN, GRB>(leds, NUM_LEDS);
  FastLED.setBrightness(50);
}

void loop() {

  for (int i = 0; i < NUM_LEDS; i++) {
    leds[i] = CHSV(100, 120, 255);
  }

FastLED.setBrightness(LED_PIN,0);

FastLED.show();
}`

Receiving error message:
no matching function for call to 'CFastLED::setBrightness(int, int)'

I've been advised that there may be an embedded spurious character.

Any ideas?

marmilicious, 2020-12-12:
This is causing the issue:
FastLED.setBrightness(LED_PIN,0);

FastLED.setBrightness only takes one value, from 0 to 255.

johnty57, 2020-12-13:
Splendid.  Thanks for helping me out.

closed 2020-12-13 by johnty57