FastLED/FastLED #1137: Initializing FastLED with non-constant pin values

sblantipodi, 2020-12-12:
Hi,
FastLED.addLeds() does not support variables as pin value.

I need to create a pre build firmware that let the user choose the digital pin to use at runtime.

I can workaround it using a switch but even if I use the FastLED.addLeds inside a switch case, it sucks up memory every time I create new case in the switch.

Is there a way to do it without wasting memory?

Thanks

kriegsman, 2020-12-14:
Hi and thanks for asking about this.  You might want to read the discussion over at https://github.com/FastLED/FastLED/issues/282, since it's very much about the same issue.

I'll also include a copy of what I just posted there:

Especially on 8-bit (AVR) microcontrollers, FastLED needs to optimize the assembly code differently depending on which which pin is being used; different pins have different ways of accessing them and FastLED tries to choose the fastest -- at compile time, so that there's no IF/THEN required at runtime.  

On faster microcontrollers, like ARM, and ESP, we could probably add the ability to dynamically set a pin number, because the cost of the IF/THEN logic at runtime is so much lower on these faster chips.  

@samguyer and I just started a conversation about how we might be able to do this in the future, and we're interested in taking a look at doing that once we get the current 'release' out the door.  So, short version of the answer is: the pin number is fixed at compile time for performance reasons.  We might be able to make it dynamic in the future, but we don't have a timeline for that.

closed 2020-12-14 by kriegsman

sblantipodi, 2020-12-14:
@kriegsman thanks for the kind answer. 
This feature should not be considered a low priority since it's not possible to build a prebuild firmware with those restrictions.

Hope you'll fix it soon.

Thanks

sblantipodi, 2020-12-14:
@kriegsman But... Suppose that I can leave with it and I want to put the addLeds function in an if statement.

if (something) Fastled.addLeds(X) else Fastled.addLeds(Y)

Is there a way to not waste memory two times?
It seems that every time I call addLeds it consumes memory, even if inside a if then else statement.

kriegsman, 2020-12-15:
What’s happening is that the compiler is generating the special code for each pin that you name, so if you name 4 pins, it generates the special code for all four, and the code size goes up somewhat. 

What microcontroller are you using? We’d only be supporting dynamic pin assignment on fast 32-bit microcontrollers such as ARM and ESP, not on 8-bit slower-clocked microcontrollers like AVR / classic Arduino — it’s just not fast enough for us to do that well there.

kriegsman, 2020-12-15:
Also, it’s generating more code and taking up more (flash) program storage space. On most microcontrollers that’s separate from RAMmemory. Just clarifying that adding more code doesn’t _necessarily_ reduce the amount of available RAM.

samguyer, 2020-12-15:
@sblantipodi What microcontroller platform are you using?

DougLeary, 2020-12-15:
I've been using FastLED for another Christmas project just now, this time using multiple strings of leds as per the instructions in the wiki. Thank you for clarifying the different ways of doing this so well! I had no trouble at all following the examples. Doing this whole project with a single controller was an interesting challenge, but honestly ESP boards are so cheap nowadays, next time a I do a similar project I will probably just use a separate ESP for each led string and have them talk over a mesh network or something. When I see people complaining about memory consumption I have to wonder how much of their time is worth the $3 it would cost to just use multiple controllers and keep everything simple. Anyway I LOVE FastLED!!!.

sblantipodi, 2020-12-15:
> What’s happening is that the compiler is generating the special code for each pin that you name, so if you name 4 pins, it generates the special code for all four, and the code size goes up [...]

Using esp8266 and Esp32

sblantipodi, 2020-12-15:
> I've been using FastLED for another Christmas project just now, this time using multiple strings of leds as per the instructions in the wiki. Thank you for clarifying the different ways of doing [...]

Just trying to create a pre build firmware, really don't want to create a firmware for each pin possible on an esp32.

sblantipodi, 2020-12-15:
> @sblantipodi What microcontroller platform are you using?

Esp32 and esp8266

samguyer, 2020-12-15:
@sblantipodi Can you tell me a little more about the hardware setup? Do you want to be able to change the pin while the system is running, or is it more of a one-time thing when it starts up? 

For ESP32, in particular, there is very little cost to having a switch statement in the setup routine that calls addLeds with different pin parameters. I factored out 90% of the common code, so it won't be duplicated. I'm also looking at a way to set the pin dynamically, but that won't be available right away.

sblantipodi, 2020-12-15:
@samguyer thanks for the answer,
I appreciate your help here and love FastLED.

I am trying to create a firmware that can be easily flashed by newbies on an ESP device.
When you flash the firmware, it automatically start an access point, user can connect to this AP via WiFi and via web browser enter the WiFi credential and the Digital PIN it want to use.

Once all the data has been entered, ESP reboot and select the right digital pin.
So yes, it's a one time thing.

If I put 8 PIN (8 addLedds) in my switch case I get this error on ESP8266:
"`.text1' will not fit in region `iram1_0_seg'"
I would not say that it consumes few memory.

sblantipodi, 2020-12-16:
@samguyer do you think that we will see those optimizations on esp8266 soon?
Why it's harder for ws2812/apa102?
Thanks

samguyer, 2020-12-16:
@sblantipodi The pin choice should be easy(ish) to add for platforms like ESP-32 and ESP-8266. We're discussing a strategy that would keep the rest of the code more or less the same (since it's important for AVR platforms). Basically, you'd be able to say "set the pin to X" dynamically after the call to addLeds.

The issue with the LED chip is that we use template instantiation to map the chip name (like WS2812) to a set of timing numbers that describe the signal. In order to bypass the template, YOU would need to tell it the timing numbers, which I don't think is reasonable. Again, the whole thing was designed for AVR, where it is really important to compile everything down to the most optimized code.

sblantipodi, 2020-12-17:
@samguyer ok thanks for the answer. I'll wait for this feature. Thanks.

Greenstreem, 2022-01-03:
Hello @samguyer, has this feature been implemented?

samguyer, 2022-01-03:
@Greenstreem Sorry, we haven't had time to work on this feature yet. But thanks for giving it a bump!