FastLED/FastLED #1138: please release a new version of fastLED so people get all the recent fixes & updates :)

ladyada, 2020-12-14:
It's been many months since the last fastLED release, @kriegsman could you perform an arduino release for the library manager? 
to do that, library.properties must be updated to, say, 3.4.0 and then tag the same version release here
https://github.com/FastLED/FastLED/releases

(I can do it for you if you give me writing access to the repo)

kriegsman, 2020-12-14:
Thank you, @ladyada- we met just this morning and made a plan for the 3.4.0 release. I think we’re within a couple of days of doing it now. Long overdue, especially with all the good contributions folks have made.

ladyada, 2020-12-14:
awwwwwsome thank yoU!

sblantipodi, 2020-12-19:
Adding a changelog to the releases could be really appreciated by users. Thanks for your work guys.

marmilicious, 2020-12-22:
@ladyada New release:  https://github.com/FastLED/FastLED/releases/tag/3.4.0

marmilicious, 2020-12-22:
@ladyada And thank you for your contributions too!

ladyada, 2020-12-22:
awesome, merry christmas!

closed 2020-12-22 by ladyada

kriegsman, 2020-12-22:
Thank you so much for your contributions, to the whole wide community, and also specifically to the FastLED project!  Merry xmas and happy new year. Or as my dad always said, _happier_ new year!