FastLED/FastLED #1142: Support for HD107S?

johndabeast7, 2020-12-17:
I was looking into HD107S strips for use with an ambilight system due to their extremely quick refresh rate. Searching on here, I found an old topic that seems to hint that they would work in place of APA102 strips, but they don't seem to be working for me. If they are not supported, can support for them be added? Appears their PWM frequency is 27 kHz and their transmission rate is 30 mHz (or 40 mHz? Seem to be seeing differing values on this). This differs from APA102s which are 20 kHz PWM and 20 mHz transmission. One of the data sheets can be found here: http://www.ledlightinghut.com/files/HD107S.pdf

Thanks for any help you're able to provide!

marmilicious, 2020-12-20:
Sounds like NS107S and HD107S and LC8823 and XT107S and WS107S are all the same!  All the same NS107S IC made by Newstar, just different names the various production factories are using when selling their strips.

https://newstarleds.wordpress.com/2019/07/17/hd107s-ns107s-with-lc8823-leds/
http://newstarleds.blogspot.com/2019/07/the-difference-of-ns107s-hd107s-lc8823.html
https://os.mbed.com/questions/86568/which-one-is-the-most-suitable-LEDs-to-r/

Data Sheet: 
https://cdn.hackaday.io/files/21293910999712/NS107S%20RGB%20LED%20Chip.pdf

johndabeast7, 2020-12-21:
This is good to know, thank you Marmilicious. Do you know if these ICs are supported or plan to be supported by FastLED? I didn't see any of those model names under the supported hardware category.

marmilicious, 2020-12-21:
@johndabeast7  Some places make it sound like it's a drop in replacement for the APA102, but if that's not working for you then something might be slightly different.  The data sheets for these ICs are notoriously bad some times.  I imagine these will eventually be added to the library as their own pixel type, but I have no timeline for that.

closed 2026-05-24 by zackees