FastLED/FastLED #1146: ESP32 -> SPI What's the minimal line size?

JLightMedia, 2020-12-22:
Hi LED masters. 
I have not typical question. What is the minimum SPI line size?
I'll try to explain my question.

I have worked with WS2812 and Arduino boards and it wasn't any problem. After that I move on ESP8266. It were some problems. More of them was with terrible flicking I used level shifter to solved them. And now I work with ESP32. Everything was OK until I found out that my  WS2812 LED strip doesn't work when SPI line so short. (My board and LED strip so close to each other, it's no necessary to create long line). Bellow, what I found out.   

1. SPI line  10-20mm -> Doesn't work at all!
2. SPI line 20-40mm -> Works but so unstable (flickering)
3. SPI line more 40mm -> Works good.

And also, in first too case to solve that problem, I connected pull-up resistor 220om. Resistor shouldn't be so big or it will not affect. The down side of that, if I soldering resistor permanently, ESP32 doesn't start. It works, only when I connect resistor after ESP32 started. 

All of thees sounds crazy, but my lamp started work good when I soldered long wire from board to LED strip.

Does anybody have a slightest idea why the short SPI line affect on LED strip?

Is it possible that it is different data voltage level? If my LED striped powered by 5v should I use signal shifter with ESP32? And is it necessary to use 330om resistor in SPI line? I sow that some LED circuit contains that.

marmilicious, 2020-12-22:
I have not heard of a short data line causing a problem before, but sounds like something is causing some sort of interference.

- Which pin on the ESP32 are you using for data?  Some pins should probably not be used.

- Which level shifter are you using?  Some are not fast enough.  The SN74HCT245 and 125 are known to work.  Wiring would be like this:  https://i.imgur.com/9AkOWg3.png

- On the level shifter, do you have any unused inputs tied to ground (or Vcc) so they are not floating?

Mostly we use github issues here for library bugs and improvement requests. If you continue to have hardware issues please ask about this on the FastLED Reddit group.  https://www.reddit.com/r/FastLED

JLightMedia, 2020-12-23:
Thank you. I'll move this issue on reddit, can you please write your answer again in new topic to continue discussion?

JLightMedia, 2020-12-23:
https://www.reddit.com/r/FastLED/comments/kip2qh/esp32_spi_whats_the_minimal_line_size/

closed 2020-12-23 by JLightMedia