FastLED/FastLED #1147: WS2813B support

Galaxarum, 2020-12-22:
Is it planned?

marmilicious, 2020-12-23:
It should already be supported.  Did you try:

```
FastLED.addLeds<WS2813, DATA_PIN, RGB>(leds, NUM_LEDS);
```

Galaxarum, 2020-12-23:
WS2813B is the clocked variant. The solution you proposed is an hack I'm currently using, which works provided I connect the clock input to ground, but I was wandering if clock support is planned in order to exploit it and getting better results (wifi used, so clock line would reduce the impact of interrupts)

marmilicious, 2020-12-23:
No, the BIN (or also labeled as BI) input is not for a Clock signals, usually labeled CKI (or CI for Clock In) and CKO (or CO for Clock Out).  BIN stands for Backup In.  It's a backup of the data line so if a pixel goes bad, the data signal can continue on around/past the failed pixel and the rest of the strip will continue to work instead of freezing or going black.  Gives a bit of piece of mind on big installations.

Data sheet:  
http://www.normandled.com/upload/201605/WS2813%20LED%20Datasheet.pdf
https://html.alldatasheet.com/html-pdf/1134580/WORLDSEMI/WS2813/2281/4/WS2813.html

On the first first pixel I feel you should wire BIN to ground (as shown in the second data sheet linked above), but some has said they just leave it unconnected.  Doesn't sound like that worked for you, so good you grounded it and got it working.

Galaxarum, 2020-12-23:
Thank you for the info! :) Marking as solved.

closed 2020-12-23 by Galaxarum