FastLED/FastLED #1148: ESP8266 is faster than ESP32 with FastLED

sblantipodi, 2020-12-23:
Hi,
I have a 500 array LED using WS2812B chipset.

With ESP8266 @ 160MHz I can call FastLED.show() 35 times a second,
with ESP32 @240MHz I can call it 26 times a second.

Why ESP8266 is faster than ESP32 with Fastled?

marmilicious, 2020-12-23:
Can you put your test code on pastebin.com or gist.github.com and share a link to it in case someone wants to try duplicating your test?

sblantipodi, 2020-12-24:
@marmilicious thanks for the answer.

this code:
https://gist.github.com/sblantipodi/033c61503f4c3e7494a89e342f1c8682

is 1FPS faster on my ESP8266 @160MHz than on my ESP32 @240MHz

isn't ESP32 supposed to be much faster than ESP8266?

samguyer, 2020-12-30:
@sblantipodi There are two factors at work here. First, the time it takes to send data to clockless strips like the WS2812 is determined almost entirely by the protocol -- for WS2812 it is 30 microseconds per pixel (30,000 nanoseconds). Second, the driver for the ESP32 is a little more complex than the one for ESP8266 in order to support parallel output (output to multiple strips at the same time). So, for one strip, you'll see a little bit of overhead on the ESP32 that might account for 1 FPS difference with the 8266. With four or more strips, the ESP32 will blow it away.

samguyer, 2020-12-30:
@sblantipodi In your example, if you can break the 500 LED strip into five 100 LED strips and attach them to different pins, you'll be able to send all that data in 1/5 the amount time on an ESP32 (i.e., you'll have 5X higher max frame rate).

samguyer, 2021-01-02:
@sblantipodi Can I close this issue?

sblantipodi, 2021-01-02:
I really appreciate your answer and help. Thank you for your work here. 

Closing it. Thanks.

closed 2021-01-02 by sblantipodi