FastLED/FastLED #1156: doneOnChannel function should check that the channel is not higher than FASTLED_RMT_MAX_CHANNELS

fiskn, 2020-12-26:
If using the builtin RMT driver and also using RMT for other purposes, the current code will panic as a non FastLED interrupt will still action the callback function and try and assign an invalid/non-existent channel to pController.

I've added a simple check at the start of the function which seems to fix this for me, but not sure if there is a better way of doing this:

if(channel>FASTLED_RMT_MAX_CHANNELS)
		return;

samguyer, 2020-12-30:
@fiskn Thank you for this issue report. Unfortunately, it made me realize that there are several other problems related to this issue that all need to be solved. I will keep you updated.

samguyer, 2020-12-30:
@fiskn I just pushed a change that should fix this problem (and a couple of other related problems). The PR is #1163

fiskn, 2020-12-30:
Hi Sam,

Thanks for taking time to look at this. I will test the new PR out on my project a bit later tonight.

I'm just going through the changes and unless I am misunderstanding them, I'm not sure if they will fix the issue I was experiencing. 

I currently have FASTLED_RMT_MAX_CHANNELS defined as a compiler define set to 1. So from what I understand FastLED will pick the first RMT channel available and use that. I've also using RMT channel 8 separately to send IR data via a IR LED.

What I believed was happening was that due to the following line 

237:  rmt_register_tx_end_callback(doneOnChannel, 0);

Any RMT interrupts FastLED related or not, will now call the doneOnChannel function.

The first line in the doneOnChannel function tries to do this:

 ESP32RMTController * pController = gOnChannel[channel];

Which if triggered by my other RMT call on channel 8 panics, as there is no FastLED controller for channel 8. Hence I used an IF statement to make sure that function exits early if the calling channel is higher than the FASTLED_RMT_MAX_CHANNELS define.

But like I said, let me check your code later as it may have resolved this in a different way.

Also did you see my other issue 1151? I'm also not 100% sure if my solution was correct, but I don't believe manually disabling the interrupts is needed when using the built in driver and stopped the issues I was seeing once I excluded that part.

samguyer, 2020-12-30:
@fiskn I see what you're saying now -- our code is fighting for control over the RMT device with the built-in driver. I'm not sure I can fix this problem quickly. My recommendation is to use the I2S version of the ESP32 driver. It doesn't use the RMT device at all, so this problem will go away. Just add this line *before* you include FastLED.h:

#define FASTLED_ESP32_I2S 1

fiskn, 2020-12-30:
@samguyer Unfortunately, the I2S is also being used for sound output, the esp32 is getting worked pretty hard in my project (https://gitlab.com/fiskn/libretag2)

For now it's working at least for my use case with the two IF statements I have added, but I'm sure there will be corner cases which they will not work for. The start of the function now looks like this with the two extra if statements:

```
void IRAM_ATTR ESP32RMTController::doneOnChannel(rmt_channel_t channel, void * arg)
{
    if(channel>FASTLED_RMT_MAX_CHANNELS)
		return;
    ESP32RMTController * pController = gOnChannel[channel];

    // -- Turn off output on the pin
    // SZG: Do I really need to do this?
    gpio_matrix_out(pController->mPin, 0x100, 0, 0);

    if (!FASTLED_RMT_BUILTIN_DRIVER) {
	// -- Turn off the interrupts
    // rmt_set_tx_intr_en(channel, false);
    // Inline the code for rmt_tx_stop, so it can be placed in IRAM
    RMT.int_ena.val &= ~(1 << (channel * 3));
    RMT.conf_ch[channel].conf1.mem_rd_rst = 1;
    RMT.conf_ch[channel].conf1.mem_rd_rst = 0;
  }
```

samguyer, 2020-12-31:
Ah, ok, so we need to get this working. I'll take a look at your fix. To be
honest, I haven't spent a lot of time testing the built in driver.

zackees, 2024-10-12:
Obsolete with the RMT5 driver. Closing.

closed 2024-10-12 by zackees