FastLED/FastLED #1159: Array of CRGBArray

jahwerx, 2020-12-29:
My Project is perhaps unique.
I have 5 LED strips (WS2811) of varying length arranged in an organic layout.

```
#define NUM_STRIPS 5 // Here are the 5 strips declared, and some declarations on the LED length below.
#define LEDS_ON_0 13
#define LEDS_ON_1 23
#define LEDS_ON_2 3
#define LEDS_ON_3 3
#define LEDS_ON_4 8
#define LEDS_TOTAL 50
```

No issues at all cycling through them normally with 5 instances of CRGBArray and lots of logic to handle the different strips:
```
  CRGBArray<LEDS_ON_0> leds0;
  CRGBArray<LEDS_ON_1> leds1;
  CRGBArray<LEDS_ON_2> leds2;
  CRGBArray<LEDS_ON_3> leds3;
  CRGBArray<LEDS_ON_4> leds4;
```

HOWEVER, my logic could be MUCH tighter if I could iterate through a collection/array of the 5 strips. I tried the following and it just won't compile. I think each of these should work, but I'm stumped!
In every case I get the error:
invalid use of template-name 'CRGBArray' without an argument list

Try to create an "Array of Arrays", using the statements above:
`CRGBArray *ledsALL[NUM_STRIPS] = {&leds0,&leds1,&leds2,&leds3,&leds4};`

Trying to do the same Create the Array from scratch:
```
CRGBArray *ledsALL[NUM_STRIPS] = {
    new CRGBArray<LEDS_ON_0> leds0,
    new CRGBArray<LEDS_ON_1> leds1,
    new CRGBArray<LEDS_ON_2> leds2,
    new CRGBArray<LEDS_ON_3> leds3,
    new CRGBArray<LEDS_ON_4> leds4    
    };
```
Any Advice would be appreciated :)

lpaolini, 2020-12-30:
Hi,

You can do this:

```
CRGBArray<LEDS_ON_0> leds0;
CRGBArray<LEDS_ON_1> leds1;
CRGBArray<LEDS_ON_2> leds2;
CRGBArray<LEDS_ON_3> leds3;
CRGBArray<LEDS_ON_4> leds4;

CRGBSet *ledsALL[] = {&leds0,&leds1,&leds2,&leds3,&leds4};

// now you can iterate over led strips like this:
for (CRGBSet *leds : ledsALL) {
    leds->fill_rainbow(0, 5);
}

```

jahwerx, 2020-12-30:
Brilliant!!! works like a champ. This is my first project after "blink", so I'm digging in deep.

kriegsman, 2021-01-04:
Hi and welcome!  For 'how do I...' sorts of questions, please check out the FastLED subreddit at https://www.reddit.com/r/FastLED/ -- that's where you can get fastest answers to questions like these.  We generally reserve github 'Issues' for bugs and improvements in the FastLED library itself.

Thanks again for using FastLED; I'm closing this Issues as solved!

closed 2021-01-04 by kriegsman