FastLED/FastLED #1160: Cannot use ESP32 pins > 32

ausculta, 2020-12-29:
Hello,

I have ESP32 devkits with 36 pins. I'm trying to use pins 35 and 36 to connect a couple of WS2812B strings but there seems to be a hardcoded limit of 32 ESP32 pins. If I try to use pin > 32 I get a compilation error as the template cannot be resolved.

I have searched in existing issues and cannot find a reference to this. So hopefully this isn't a duplicate.

Thank you.

Kamal-Sonani, 2020-12-30:
Just check this post
https://esp32.com/viewtopic.php?t=2803

ausculta, 2020-12-30:
Thank you. As far as I know from the diagrams I've seen (DOIT DEVKIT V1 - 36 pins), GPIO 32-36  and GPIO 39 are all available although I have several different diagram versions. One suggests that GPIO34-36 may be digital input only (rather than digital input/output + ADC channels) on some boards. GPIO33 is definitely usable as digital input/output, though.

samguyer, 2020-12-30:
@ausculta GPIO 34-39 are all input-only as defined by the ESP32 microprocessor itself. You should be able to use GPIO 32 and 33, though (I don't fully understand the issues with the 32K timing component).

Kamal-Sonani, 2020-12-30:
Check that 32 & 33 pin connected with rtc or not?

ausculta, 2021-01-11:
The diagrams I had were wrong. It took me a while to work out what was and wasn't connected. I also needed to remove connections to GPIO6-11 that I thought were to other GPIO pins based on the diagrams I had. 
I noticed the special handling for GPIO32+ in the code and wondered why that was. GPIO33 compiles and should work but I've rewired my prototype to use different GPIO and leave GPIO32+ for input so I can't confirm that 100%.
Now that I know which pins are connected to the ESP32, it's a lot easier to manage!
Thanks for the help.

closed 2021-01-11 by ausculta