FastLED/FastLED #1162: ESP8266 (NodeMCU) + WS2811 won't work with FastLED

toteto, 2020-12-30:
I have WS2811 LED strip of 50 "pixels" (each pixel 3 LEDs) powered by 12V supply all wired up to share ground with the ESP8266.

Using the examples from NeoPixel it works flawlessly using `NEO_GRB + NEO_KHZ800` or `NEO_GRB + NEO_KHZ400`. 

When switching to example from FastLED (RGBCalibrate) can't get it to work. Only modification I do is that i change the `#define DATA_PIN 1` and `#define NUM_LEDS 50`. Uncommenting the initializations of adding the LEDs to the `FastLED` library yields no results. I have tried every single configuration, this incudes the one for `WS2811`, `WS2811_400` and `WS2812`.

Doing some debugging with printing in serial monitor, i can confirm that the `loop()` method is executed each second.

Tried downgrading the library, but the few i have tried didn't yield results too.

This is some error message I get when compiling, don't think is related to the issue I have.

```
Arduino: 1.8.13 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 460800"

In file included from /var/folders/dn/_8yjdk6x68v608xn0b_h2c2w0000gn/T/arduino_modified_sketch_551053/RGBCalibrate.ino:1:0:
/Users/tote/Documents/Arduino/libraries/FastLED/src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
 #    pragma message "FastLED version 3.004.000"
                     ^
In file included from /Users/tote/Documents/Arduino/libraries/FastLED/src/FastLED.h:65:0,
                 from /var/folders/dn/_8yjdk6x68v608xn0b_h2c2w0000gn/T/arduino_modified_sketch_551053/RGBCalibrate.ino:1:
/Users/tote/Documents/Arduino/libraries/FastLED/src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
```

Is there any info that i can provide for resolving of the issue?

samguyer, 2020-12-30:
@toteto Those warnings are just for information -- there is no compilation problem. Can you share your code on gist or github?

samguyer, 2020-12-30:
Also: what is the behavior? Is it just that you're not seeing anything on the LED strip?

toteto, 2020-12-30:
@samguyer 

The result I get on the led is whatever previous state they were in. For example when i used the NeoPixel example with rainbow effect, and i uploaded the FastLED example, the LEDs got stuck in their latest "rainbow" state. If the LEDs got powered off and on, they would stay white.

Here is the latest example with my modifications i have tried: https://gist.github.com/toteto/52d6b5a9e785ee3fc1500c7dedf7b662

Thanks for your time.

samguyer, 2020-12-30:
@toteto Hmmm. That is a very basic configuration that many people have used successfully. I'm wondering if maybe the pin names are the problem: "D1" is not necessarily the same as "1". For example, on the ESP12E, it looks like pin "1" is the reset pin, while IO1 is actually pin 22.

toteto, 2020-12-31:
@samguyer I have used this pinout guide

[image]

I am connecting to the TX pin that is GPIO1. I think that this pin is OK since i have used it successfully with NeoPixel library.

I have stumbled uppon https://github.com/FastLED/FastLED/wiki/ESP8266-notes, will try few things with it and see how it plays out.

toteto, 2020-12-31:
Yup, that was it, i have been using GPIO pin layout when it was required to use NodeMCU pin layout. Using `DATA_PIN 5` and connecting to LED strip to `D5` it is working. 

@samguyer thanks again for your assistance. I am closing this issue.

closed 2020-12-31 by toteto

samguyer, 2020-12-31:
@toteto Great! Glad to hear it.
It's frustrating that the pin layouts for these boards is so confusing.