FastLED/FastLED #1164: ESP32 SPI output like a sine wav

DexterHill0, 2021-01-02:
I've been having a really weird issue with my ESP32 and controlling my WS2811 (SM16703P are the chips I have) LED strips. When I try to set them to any colour they'll just show random colours of random brightness or just the wrong colour completely. I thought this was an issue with the fact that I was trying to combine FastLED and EspAlexa. I decided I'd wait to open an issue until I'd finished my app that communicates via bluetooth instead. I was hopeful this would fix it but the issue of weird colours remained. I've done lots of debugging my latest being gettting a basic understanding of the SPI protocol and how the WS2811 chips actually receive and process the data.
Looking at the datasheet for my version of the chips it shows that data should be sent in sections of 24 bits, 8 bits representing red, 8 bits for blue and 8 bits for green. This makes sense because 8 bits is what you need for 255. I also found out that there is a minimum-low and a minimum-high voltage that needs to be reached in order for the chip to consider it low / high. I thinks its usually like (0.7 * logic level) for the minimum high and something else for the low.

I decided to use my (cheap) oscilloscope on the output of my UNO to see what the data is. I used this simple script to set them to white:
```
#define NUM_LEDS 45
#define DATA_PIN 7 //I used pin 13 on the ESP

CRGB leds[NUM_LEDS];

void setup() {
    delay(2000);
    FastLED.addLeds<WS2811, DATA_PIN, RGB>(leds, NUM_LEDS);
}

void loop() {
    fill_solid(leds, NUM_LEDS, CRGB(255,255,255));
    FastLED.show();
    delay(2);
}
```
On the oscilloscope this looks like:
[image]
From what I understand this is the right output (it also lit up the lights white so it must be).

The ESP on the other hand:
[image]
Even with the sec/div being slightly different it's very clear that the output of the ESP follows some kind of sine / triangle wave. Correct me if I'm wrong, but this is not supposed to happen. The is causing the lights to display wrong because some of the highs are not reaching the minimum high voltage and so are considered by the chip to be "low" instead. 

If this is not supposed to happen is my esp broken or is there something I have enabled / need to disable to get these working?

samguyer, 2021-01-03:
@DexterHill0 Which version of FastLED are you using? We don't use SPI at all on the ESP32 (although there is some work being done on it, but only for clocked chips like the APA102). Can you tell us more about your hardware setup? One issue you might be running into is that the ESP32 is a 3.3V device, whereas the UNO and the WS2812 LEDs are both 5V devices. So any voltage drop along the data line makes the signal too low for the WS2812 to read.

samguyer, 2021-01-03:
Some people have found that they need a logic-level shifter to get the signal coming off the ESP32 pin from 3.3V up to 5V.

marmilicious, 2021-01-03:
@DexterHill0  Just to confirm, your pattern displays correctly on the LEDs when run from the UNO?  How are you powering the LEDs?

When you switched to the ESP32, do you have a common ground between the controller and the LED strip?  
When you get a level shifter make sure you get one that's fast enough.  Here's two that are known to work and wiring would be like this:  
https://i.imgur.com/9AkOWg3.png

DexterHill0, 2021-01-03:
@samguyer I tried both my main LED strips (24V) and a random strip I had lying around (5V). Both had and external supply with a common ground to the ESP. I used data pin 13 on the ESP and 7 on the UNO. I did also try a logic level shifter although it didn't seem to make any difference. (The ones I used are just these [transistor](https://www.ebay.co.uk/i/383859589132) based ones)

@marmilicious I've bought a BLE module for the UNO so I can make sure it's not an issue with my code and that a different MCU will produce correct results. That should be arriving today so once it has I will say if the UNO works or not then.

I do want to add that I tested with the DemoReel100 example and my own that cycled through RGB and all combination of RGB and the led strips lit up correctly (at 3.3V). The moment the script becomes any more complicated (there's more than just FastLED imported) it seems to break down.
I'll make sure to record a video of them and post it here.

DexterHill0, 2021-01-03:
Ok so I did a few more tests and recorded some videos (they're kind of bad because the lights swamp the camera). I used the Fire2012 example for the tests.
Here's a video of the lights without a level shifter:
[https://i.imgur.com/tm93SIY.mp4](https://i.imgur.com/tm93SIY.mp4) (too big for github)
Here's with the level shifter and all unused pins tied to ground:
[https://i.imgur.com/94FPIDe.mp4](https://i.imgur.com/94FPIDe.mp4)
It's better in that there's no "floating noise" and just totally wrong colours but comparing it to the video in the comments of the example, my lights still look wrong (I don't think they should be filled with white).

I also tried with my original Bluetooth code, controlling it with my own app (still with the level shifter). I was repeatedly tapping on **the same colour** (RGB: 139, 160, 200).
[https://i.imgur.com/DTAQXe7.mp4](https://i.imgur.com/DTAQXe7.mp4)
I want to stress that this was me clicking on the same colour. I was not changing the colour at all.

Once I receive my BLE module I will test with that too

DexterHill0, 2021-01-04:
It got a little bit more complicated once I realised that I had been coding everything using C++ STL which AVR boards like the uno do not support. I'm gonna need to make big changes to my program so it compiles for the uno.
I'm also busy with school but I will do my best to test with the bluetooth module as soon as I can.

samguyer, 2021-01-04:
@DexterHill0 I'm looking your oscilloscope image, and I'm noticing that V max is only 2.23V, which is way too low to drive the LEDs (even 3.3V is barely enough). Is there something else in your circuit that's pulling the voltage down? Can you measure VCC on the power rails?

kriegsman, 2021-01-04:
I agree with @samguyer here: this looks a lot like a power problem.  The _voltage_ coming out of the MCU pins should always be around either 0v or 3.3v (Vcc).   If it's significantly lower, something is wrong with the chip.  If Vcc is dropping much below 3.3v, then perhaps a bigger power supply is warranted to handle the current draw without a voltage drop?

DexterHill0, 2021-01-04:
@samguyer That _was_ without the level shifter attached. The power rails on the ESP measure 4.8V and 3.2V. I would say maybe because there's so many chips it doesn't have the current or something but if that was the case it should have worked on the short 20 ish led strip I tried yet it didnt'.

yesyesuk, 2021-01-04:
What does the output signal look like if you connect your scope to it but nothing else (disconnect LED strip, level shifters, ...)?

DexterHill0, 2021-01-05:
@yesyesuk The original oscilloscope picture is with nothing connected.

I just managed to get the uno working with the BLE module. I tested it on my short led strip because I could not get it to work with my main one (the one in the videos). I manually sent the data to it and it displayed all colours I sent on lights perfectly, unlike the esp.
My next plan of attack, I guess, is to just get another esp and compare my one to the new one.

DexterHill0, 2021-01-16:
Unfortunately I have made no progress forward, but much backwards. I'm now unable to even get the leds to change colour at this point. They stay white, no matter what I try and tell them to do (white being the default colour when they are turned on). I've tried 2 different ESPs with and without a level shifter, two different UNOs, many different configurations (ie. different pins, different chip types) and I got no response from the leds at all.
I think what both samguyer mentioned about voltage is probably a likely cause although I would expect the UNOs to work because they are 5V logic level, if that was the case. Maybe I should try a different logic level converter. 

I believe this is not a FastLED issue, though, so I'll close this for now. Thanks for all the help!

closed 2021-01-16 by DexterHill0