FastLED/FastLED #1165: Unable to change I2S_DEVICE from 0 to 1 on ESP32

drewandre, 2021-01-03:
The expanded ESP32 support is amazing -- using the I2S bus has completely eliminated any glitching that my projects have had even with wifi/bluetooth integration. I'm having one issue though where if I `#define I2S_DEVICE 1`, there doesn't seem to be any output to my WS2812B leds. I2S_DEVICE 0 works fine. One of my custom boards has an SGLT5000 running on I2S 0 so I can't change that (since master clock is running on GPIO 0), so that's why I'm looking to use I2S device 1. I posted [a similar issue](https://github.com/bbulkow/FastLED-idf/issues/21) a while back on @bbulkow's FastLED-idf thinking maybe this issue was unique to his fork, but I'm seeing the same behavior on the latest FastLED master.

I'm using the esp-idf v4.0 build environment with arduino-esp32 built as a component. I've never used the Arduino IDE or build environment for any FastLED/esp32 projects. I imagine this issue would be present in both build environments? I can get the Arduino build environment up and running if anyone would like me to try to reproduce this issue there. I've so far been able to reproduce this using esp-idf v3.3, v4.0, and v4.2, as well as when using the FastLED-idf fork so I think it should be relatively simple to reproduce in different environments.

Are there any GPIO pin considerations I may have overlooked? I assumed no because I see this note about routing from the default I2S pin to whichever pin you're using in clockless_i2s_esp32.h:
```
// -- Set up the pin We have to do two things: configure the
//    actual GPIO pin, and route the output from the default
//    pin (determined by the I2S device) to the pin we
//    want. We compute the default pin using the index of this
//    controller in the array. This order is crucial because
//    the bits must go into the DMA buffer in the same order.

mPin = gpio_num_t(DATA_PIN);
...
```

Steps to reproduce:
- Configure machine for esp-idf release/v4.0, run `menuconfig` with default settings, etc. There are no special sdk configurations needed to reproduce and I don't have auto-start arduino enabled. But like I said above, this should likely be reproducible in other environments/configurations.
- Compile and flash [this simple example project](https://gist.github.com/drewandre/81016b3da5c5a91ae5596e7f213c960c) and ensure the rainbow animation runs correctly
- Comment back in `#define I2S_DEVICE 1` towards the top of main.cpp
- Compile and flash again and note that the rainbow animation no longer displays

Versions:
- ESP-IDF 4.0 (release/4.0, commit [463a9d8](https://github.com/espressif/esp-idf/commit/463a9d8b7f9af8205222b80707f9bdbba7c530e1))
- arduino-esp32 (v1.0.4, commit [4638628](https://github.com/espressif/arduino-esp32/commit/4638628873a061c36faffebe4d146d13f960076d))
- FastLED master (master, commit [41652f2](https://github.com/FastLED/FastLED/commit/41652f2045ec9df7d5faf5b35efac280e13837f5))

samguyer, 2021-01-15:
@drewandre I had a brief discussion with the developer who originally wrote the I2S code for FastLED, and he says that I2S device 1 has different properties and cannot support more than 8-way parallelism. He's going to look into it more, though.

drewandre, 2021-01-26:
@samguyer thanks for the info. I'll stick to i2s port 0 for now. just out of curiosity do you know how specifically the properties differ? I don't see anything about the two ports having unique properties in [the espressif i2s docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html).

drewandre, 2021-02-16:
Actually, I do see [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#_CPPv410i2s_mode_t) in the espressif I2S docs that "PDM and built-in DAC functions are only supported on I2S0 for current ESP32 chip". That's the only difference between I2S 0 and I2S 1 that I can find.

Spencatro, 2021-12-02:
I believe I found what they meant, in the [technical reference manual](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf)--page 303, section 12.1, note 1: "... (note that I2S1 does not support 24-bit width)."

Bummer!

StefanRvO, 2021-12-05:
I guess that mean that it should be possible to get I2S1 working, just with fewer parallel channels.

I currently have a board that uses I2S0 for an audio ADC, so i can't use that either.

I really just need maximum 2 LED channels, so it seems like it should in theory be fine with I2S1, "just" need to adjust  the code a bit. I'm not really familiar with the code at all, but i might attempt to give it a try over Christmas and see if i can get it working.

@samguyer if you have any hints on what is needed it would be appreciated, but i guess it would mostly be stuff in i2sInit() to set it to use 16 or 8 bit mode instead of 24, and maybe a bit of mapping the pixel into the buffers.