FastLED/FastLED #1169: ESP32 - : No hardware SPI pins defined.

kooscode, 2021-01-07:
This is false, the ESP32 has hardware pins AND I am specifying the hardware SPI MOSI pin...

my test code:
```
CRGB leds[NUM_LEDS];
void setup() 
{
  LEDS.addLeds<WS2811, GPIO_NUM_23, GRB>(leds,NUM_LEDS);
  LEDS.setBrightness(BRIGHT_LEDS);
}
```

WARNINGS:

```In file included from .pio/libdeps/esp32dev/FastLED/src/FastLED.h:65:0,
                 from src/main.cpp:3:
.pio/libdeps/esp32dev/FastLED/src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"```

samguyer, 2021-01-15:
@kooscode This message is just a warning, and all it's telling you is that FastLED isn't using the SPI protocol to send the bits. We actually never use SPI to send the signal for clockless chips like the WS281x family. We should really just get rid of the warning.

marmilicious, 2021-01-15:
I feel like we've been asked about that message _many_ times over the years.  Unless it's needs to be shown in some specific case I think removing it would be a fine idea.

samguyer, 2021-01-15:
I made an issue for myself to make that message go away.

kooscode, 2021-01-17:
> I made an issue for myself to make that message go away.

Thank you !!

ByronAP, 2022-01-04:
Same issue as https://github.com/FastLED/FastLED/pull/1047#issuecomment-951326963 here when using ESP32 S2

pr0hv0st, 2022-12-14:
https://stackoverflow.com/questions/8071579/hide-gcc-warning-set-but-not-used

samguyer, 2023-08-30:
I don't have that commit information handy. But I think it's a good call. We still get a lot of questions about the warnings.

versamodule, 2023-09-14:
Using the ESP32-S3 with PlatformIO.  Does not even compile for me with errors about SPI pins. 

```

Building in release mode
Compiling .pio\build\m5stack-stamps3\src\main.cpp.o
Compiling .pio\build\m5stack-stamps3\lib3db\SPI\SPI.cpp.o
C:/Users/Jim/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp: In member function 'void SPIClass::begin(int8_t, int8_t, int8_t, int8_t)':
C:/Users/Jim/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:87:37: error: 'SCK' was not declared in this scope
         _sck = (_spi_num == FSPI) ? SCK : -1;
                                     ^~~
C:/Users/Jim/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:87:37: note: suggested alternative: 'SCL'
         _sck = (_spi_num == FSPI) ? SCK : -1;
                                     ^~~
                                     SCL
C:/Users/Jim/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:88:38: error: 'MISO' was not declared in this scope
         _miso = (_spi_num == FSPI) ? MISO : -1;
                                      ^~~~
C:/Users/Jim/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:89:38: error: 'MOSI' was not declared in this scope
         _mosi = (_spi_num == FSPI) ? MOSI : -1;
                                      ^~~~
C:/Users/Jim/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:90:36: error: 'SS' was not declared in this scope
         _ss = (_spi_num == FSPI) ? SS : -1;
                                    ^~
C:/Users/Jim/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:90:36: note: suggested alternative: 'PS'
         _ss = (_spi_num == FSPI) ? SS : -1;
                                    ^~
                                    PS
*** [.pio\build\m5stack-stamps3\lib3db\SPI\SPI.cpp.o] Error 1
In file included from .pio/libdeps/m5stack-stamps3/FastLED/src/FastLED.h:75,
                 from src/main.cpp:5:
.pio/libdeps/m5stack-stamps3/FastLED/src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
==================================================================================== [FAILED] Took 2.79 seconds ====================================================================================
 *  The terminal process "C:\Users\Jim\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

glpalma, 2023-09-24:
> Using the ESP32-S3 with PlatformIO. Does not even compile for me with errors about SPI pins.
>
> ```
>
> Building in release mode
> Compiling .pio\build\m5stack-stamps3\src\main.cpp.o
> Compiling [...]

I have the exact same error. I saw on another post (which I can't find right now) that defining `FASTLED_ALL_PINS_HARDWARE_SPI` before including the library could solve this issue, but the errors remain the same. 

The code I'm trying to use is the one found in [M5Stack's StampS3 repo](https://github.com/m5stack/STAMP-S3/blob/main/examples/Led/Led.ino): 

`/*
*******************************************************************************
* Copyright (c) 2022 by M5Stack
*                  Equipped with StampS3 sample source code
*                          配套  StampS3 示例源代码
* Visit for more information: https://docs.m5stack.com/en/core/stamps3
* 获取更多资料请访问: https://docs.m5stack.com/zh_CN/core/stamps3
*
* Describe: LED Show example.  LED展示示例
* Date: 2023/1/11
*******************************************************************************
Press button to change LED status
按下按键切换LED状态.
*/
#include <Arduino.h>
#include <FastLED.h>

#define PIN_BUTTON 0
#define PIN_LED    21
#define NUM_LEDS   1

CRGB leds[NUM_LEDS];
uint8_t led_ih             = 0;
uint8_t led_status         = 0;
String led_status_string[] = {"Rainbow", "Red", "Green", "Blue"};

/* After StampS3 is started or reset
   the program in the setUp () function will be run, and this part will only be
   run once.
   在StampS3启动或者复位后，即会开始执行setup()函数中的程序，该部分只会执行一次。
*/
void setup() {
    USBSerial.begin(115200);
    USBSerial.println("StampS3 demo!");

    pinMode(PIN_BUTTON, INPUT);

    FastLED.addLeds<WS2812, PIN_LED, GRB>(leds, NUM_LEDS);
}

/* After the program in setup() runs, it runs the program in loop()
  The loop() function is an infinite loop in which the program runs repeatedly
  在setup()函数中的程序执行完后，会接着执行loop()函数中的程序
  loop()函数是一个死循环，其中的程序会不断的重复运行
*/
void loop() {
    switch (led_status) {
        case 0:
            leds[0] = CHSV(led_ih, 255, 255);
            break;
        case 1:
            leds[0] = CRGB::Red;
            break;
        case 2:
            leds[0] = CRGB::Green;
            break;
        case 3:
            leds[0] = CRGB::Blue;
            break;
        default:
            break;
    }
    FastLED.show();
    led_ih++;
    delay(15);

    if (!digitalRead(PIN_BUTTON)) {
        delay(5);
        if (!digitalRead(PIN_BUTTON)) {
            led_status++;
            if (led_status > 3) led_status = 0;
            while (!digitalRead(PIN_BUTTON))
                ;
            USBSerial.print("LED status updated: ");
            USBSerial.println(led_status_string[led_status]);
        }
    }
}`
I am trying to flash this using PlatformIO.

huster-songtao, 2023-11-07:
Using the XIAO_ESP32 C3 with Arduino.

////////////////////////////////////////////////////////////////////////////////
// Arduino SmartLed.ino
// Copyright (C) 2023 Bot <st@bot.com>
// 
// IDE:
//   Arduino 2.2.1
// Platform:
//   esp32 3.0.0-alpha2 - https://github.com/espressif/arduino-esp32
// Board:
//   XIAO_ESP32 C3
// Board Settings:
//   USB CDC On Boot: Enabled
//   CPU Frequency: 160MHz(WiFi)
//   Core Debug Level: None
//   Erase All Flash Before Sketch Upload: Disabled
//   Flash Frequency: 80MHz
//   Flash Mode: QIO
//   Flash Size: 4MB(32Mb)
//   Partition Scheme: Default 4MB with spiffs(1.2MB APP/1.5MB SPIFFS)
//   Upload Speed: 921600
// Libraries:
//   elapsedMillis 1.0.6 - https://github.com/pfeerick/elapsedMillis

#include "FastLED.h" 
 
#define NUM_LEDS 364 
#define DATA_PIN 10 
#define LED_TYPE WS2812
#define COLOR_ORDER RGB   
 
uint8_t max_bright = 128;   
 
CRGB leds[NUM_LEDS];  

void setup() {
  // put your setup code here, to run once:
  LEDS.addLeds<LED_TYPE, DATA_PIN, COLOR_ORDER>(leds, NUM_LEDS); 
  FastLED.setBrightness(max_bright);       
}

void loop() {
  // put your main code here, to run repeatedly:
  leds[0] = CRGB::Red;      
  FastLED.show();           
  delay(50);               

  leds[0] = CRGB::Black;
  FastLED.show();
  delay(1000);
}

The code above throw exception:
In file included from d:\Arduino\Sketchbook\libraries\FastLED\src/FastLED.h:75,
                 from D:\Arduino_SmartLed\Arduino_SmartLed.ino:24:
d:\Arduino\Sketchbook\libraries\FastLED\src/fastspi.h:157:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  157 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\Sketchbook\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In member function 'void ESP32RMTController::startOnChannel(int)':
d:\Arduino\Sketchbook\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:266:29: error: 'RMTMEM' was not declared in this scope
  266 |         mRMT_mem_start = & (RMTMEM.chan[mRMT_channel].data32[0].val);
      |                             ^~~~~~
d:\Arduino\Sketchbook\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::doneOnChannel(rmt_channel_t, void*)':
d:\Arduino\Sketchbook\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:347:5: error: 'gpio_matrix_out' was not declared in this scope; did you mean 'gpio_iomux_out'?
  347 |     gpio_matrix_out(pController->mPin, 0x100, 0, 0);
      |     ^~~~~~~~~~~~~~~
      |     gpio_iomux_out

exit status 1

Compilation error: exit status 1

johnwargo, 2023-11-12:
@huster-songtao you should make a new issue for your problem as this thread is all about bit bang output, not the specific error you're having.

huster-songtao, 2023-11-13:
> @huster-songtao you should make a new issue for your problem as this thread is all about bit bang output, not the specific error you're having.

> No hardware SPI pins defined.

Actually, it's a same problem

huster-songtao, 2023-11-13:
> No, it's not. You're getting the same WARNING, telling you tha you have no SPI pins defined, but you're getting a different ERROR, which is why it quit compiling. Two different problems.
> […](#)

It looks like FastLED is affected by a bug: [FastLED/FastLED#1553](https://github.com/FastLED/FastLED/issues/1553)
I am using [Seeed Studio XIAO ESP32C3](https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html) and 3.0.0-alpha2 of the Arduino for esp32 by Espressif Systems.

Do I need to delete my comments? Or leave it for later generations to read?

marmilicious, 2023-11-13:
> Do I need to delete my comments? Or leave it for later generations to read?

Leaving it is fine.

nicelight, 2023-12-11:
> This is false, the ESP32 has hardware pins AND I am specifying the hardware SPI MOSI pin...
> 
> my test code:
> 
> ```
> CRGB leds[NUM_LEDS];
> void setup() 
> {
>   LEDS.addLeds<WS2811, GPIO_NUM_23, GRB>(leds,NUM_LEDS);
>   LEDS.setBrightness(BRIGHT_LEDS);
> }
> ```
> 
> WARNINGS:
> 
> ```
>                  from src/main.cpp:3:
> .pio/libdeps/esp32dev/FastLED/src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
>  #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"```
> ```

Hi. There is the way how to deal with:
Go to file 
~\Documents\Arduino\libraries\FastLED\src\fastspi.h
delete lines 142, 149. 
don't delete line 150: #endif
save file. 
Profit

huster-songtao, 2023-12-11:
@nicelight pls check your esp32, do not use 3.0.0-alpha, you can use 2.0.14. It work well for me.

```
#define FASTLED_INTERNAL //remove annoying pragma messages
#include <FastLED.h>
```

you  can add code : #define FASTLED_INTERNAL //remove annoying pragma messages

zGigit, 2024-02-17:
> Hi. There is the way how to deal with: Go to file ~\Documents\Arduino\libraries\FastLED\src\fastspi.h delete lines 142 -149. don't delete line 150: #endif save file. Profit

I don't know why this warning message grinds my gears so, but here we are. 

I found nicelight's solution muy bweño callieyentay indeed. It got me 99% to this solution, so credit where due.

Except, either the code has shifted since the time of the post, or it is possible the block they specifically removed worked to stop the message for their specific hardware.  Because, while it didn't fix the issue for me, it also didn't break the code completely for me either.

But the following instructions should fix it, irrespective of shifting code, or whichever your chip of choice:

STEP 1:
In ...\Arduino\libraries\FastLED\src\fastspi.h, find the block pragma message itself with cmnd F; and search for  "No hardware SPI pins defined.." yada yada.. 

As of v 3.6.0, the blocks to comment out are:
   a) lines 154 thru 161
       then skip a few lines that are already commented out, as well as one "#endif", 
       and then comment out: 
  b) 169 thru 176.

Which altogether should leave you with:

`/*  #else
#  if !defined(FASTLED_INTERNAL) && !defined(FASTLED_ALL_PINS_HARDWARE_SPI)
#    ifdef FASTLED_HAS_PRAGMA_MESSAGE
#      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
#    else
#      warning "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
#    endif
#  endif  */
#endif

// #if defined(USART_DATA) && defined(USART_CLOCK)
// template<uint32_t SPI_SPEED>
// class AVRSPIOutput<USART_DATA, USART_CLOCK, SPI_SPEED> : public AVRUSARTSPIOutput<USART_DATA, USART_CLOCK, SPI_SPEED> {};
// #endif

/*  #else
#  if !defined(FASTLED_INTERNAL) && !defined(FASTLED_ALL_PINS_HARDWARE_SPI)
#    ifdef FASTLED_HAS_PRAGMA_MESSAGE
#      pragma message "Forcing software SPI - no hardware SPI for you!"
#    else
#      warning "Forcing software SPI - no hardware SPI for you!"
#    endif
#  endif  */
#endif`

Feel free to delete instead of comment out the code, I just prefer the latter (in case one day, overcome with Stockholm Syndrome I realize I've grown to miss the way my heart would palpitate as time fell away, leaving only me and the red text block of judgement that might as well have read: "your code is nothing.  and it will never compile to anything."

dragetd, 2024-04-10:
Any news on this?

While I am actually not how this warning can be avoided since it seems to be at macro level - I guess you do not know that I want to control a ws2812b at that time - the message could be at least updated to be less confusing…?

`No hardware SPI pins defined.  All SPI access will default to bitbanged output` -> `No hardware SPI pins defined.  All SPI access will default to bitbanged output. If you are using a clockless LED like WS281x, ???, etc., you can safely ignore this warning`

I also tried adding:
```
#define FASTLED_ALL_PINS_HARDWARE_SPI
#define FASTLED_ESP32_SPI_BUS HSPI
```

Which gives a less confusing message at least.

BCsabaEngine, 2024-04-18:
> Any news on this?
>
> While I am actually not how this warning can be avoided since it seems to be at macro level - I guess you do not know that I want to control a ws2812b at that time - the message [...]

I have deleted the two top lines from fastspi_esp32.h :(

huster-songtao, 2024-05-10:
How to define SPI pins?

geekasylum, 2024-07-29:
Three years on, and we still have this problem.

In PlatformIO this "warning" causes a fatal compile error and ends compilarion (at least for me and my esp32-s3 setup).

For those following along behind me, the work-around was posted above by @huster-songtao and is repeated here for efficiency:

```
#define FASTLED_INTERNAL //remove annoying pragma messages
#include <FastLED.h>
```

This causes FastLED to skip over this code and resolvced the issue for me, so I could move on to other things.

coocy, 2024-08-01:
> Three years on, and we still have this problem.
> 
> In PlatformIO this "warning" causes a fatal compile error and ends compilarion (at least for me and my esp32-s3 setup).
> 
> For those following along behind me, the work-around was posted above by @huster-songtao and is repeated here for efficiency:
> 
> ```
> #define FASTLED_INTERNAL //remove annoying pragma messages
> #include <FastLED.h>
> ```
> 
> This causes FastLED to skip over this code and appears to have resolved the issue for me, so I could move on to other things.

Thanks, this worked for me.

closed 2024-08-17 by zackees

geekasylum, 2024-09-01:
> Yes, IMO this warning causes more problems than it solves, but your build should treat warnings as warnings and errors as errors, especially when you don't have right control of the code and build environment. A complier update may change warnings, for example. Building in a new OS or a new cross build would result in breakage. Both sides could improve here, IMO.

In my case, I simply need to get a job done. I write my code and make it work and that is all I should need to do. I dont have the time, or the will to spend time leaning _how_ a 'build environment' does its job... I only care that it does it.  Anything beyond the scope of the code that I have written is maintained by others and is not my concern.  I don't need to know the physics of how a hammer works in order to drive a nail. It is simply a tool that I use to get a job done... as is a build environment.

To that end, I no longer use Platformio. It had a few other issues besides this one. I went back to the Arduino IDE, and reworked the project without depending on this library, since updates to various other things made the example code I was referring to, no longer work, so I came up with something else.

The point is, there are others like me. We should not have to delve deep into the inner workings of our compilers just to make them 'compile' third party code.

zackees, 2024-09-06:
I just tried to repro this case with the supplied code and everything now works fine.