FastLED/FastLED #1175: Crazy flashing with 3535 LEDs (SK6812 mini), FastLED, Adafruit M0 Express, latest LED version in China

shaunmavronicolas, 2021-01-13:
Hello everyone.

I would so appreciate some assistance as we surely cannot be the only ones who have or will run into this problem below.

We just received newly assembled 70 LED 3535 matrixes (SK6812 mini). The actual LED spec has not changed since the last version we used of these LEDs in April 2019. However, In August 2019, I was told it was EOL for the chip inside the LED and a new version was all that was available, this upon reordering from PCBWay.

So we went ahead and had our boards made. We just received them. They all fail with mad flashing and random pixels and colours.

Before I go one, we are running 3.3.3 version of FastLED and using an Adafruit M0 Express. 

If we drop the voltage to the LEDs to below 4.8V, the flashing issue stops and all works as normal. The boards we had made with the older version of LED chip in April 2019 have no issues what so ever on our hardware/firmware, switch out the panels to the new ones and the problem is there (100% fail).

We contacted the LED manufacturer, Dongguan OPSCO Optoelectronics Co.,ltd. After ruling out LED damage due to heat, after pictures, we were asked to do this low level stuff which we cannot... "Our engineer need you to test the data by MCU when you sent the "0"code . The voltage high then the temperature is high it may impact the signal."

In the spec sheet on page 7, there is a breakdown of data transmission time.
[SK68XX MINI-X REV.08 EN.pdf](https://github.com/FastLED/FastLED/files/5810103/SK68XX.MINI-X.REV.08.EN.pdf)

I am wondering if any hardcore folk know what timing is used in the FastLED library. Is it within the spec?

Has anyone had this experience and resolved it. We can change our hardware an incorporate a regulator to the panels, but this seems to be masking a real problem. These LEDs are supposed to work up to 5.5V. At 5.2V all the boards fail, the unstable flashing, bright white, random colours, at 4.8V the chaos stops.

Would be so grateful to anyone who has ideas or thoughts...

Thank you so much.
Shaun.

203Null, 2021-01-14:
I had something similar on my hardware too if it's powered with my iPad Pro
LC8812B (Basically SK6812) on STM32F103 
If needed I will attach a video example

shaunmavronicolas, 2021-01-14:
OK, I got an update from an engineer at the LED company. Does anyone have access to a digital oscilloscope that is currently using an Arduino and running the FastLED library, that could actually measure what the timing actually is, as outlined below?  

What as proposed here, references the above PDF for details, page 7, on the timing requirements, but what's important right now is what the actual timing is for the code "0" and code "1" bits of data are and how it differs from what the manufacturer is requiring.

Below is what was written along with pictures...

"Please measure the width of  "0" code and  "1" code with an oscilloscope. Please modify the  parameters in the file of "clockless_arm_sarn.h". Reference the data of 1 code waveform, picture attached..."

[image]

[image]

samguyer, 2021-01-15:
It sounds more like an electrical problem than a software/signal generation problem. If FastLED was generating the wrong timing, the voltage to the LEDs wouldn't make any difference. Are you using a logic level shifter for the data pin? The Adafruit Feather M0 express is 3.3V MCU, so technically you need a level shifter to get the signal up to 5V for the SK6812. That might explain why lowering the voltage on the LEDs helps -- it closes the voltage gap enough that the signal works.

shaunmavronicolas, 2021-01-15:
Hi Sam. Yes, there is a level shifter. 

Just last night, we may have uncovered something. We've determined something in our code is affecting the actual data timing, CPU loading of sorts. What exactly, we don't know right now. Will update once we know more.

Cheers.

marmilicious, 2021-01-15:
@shaunmavronicolas That's good you're using a level shifter.  Which one are you using? (Some are not fast enough which is why I'm curious to ask.)  
If the issue is with your code I hope you can sort that out, and let us know what was wrong.

shaunmavronicolas, 2021-01-15:
Sure, we're using the 74HCT1G125

shaunmavronicolas, 2021-01-20:
Hello everyone, looks to be solved, if not I will update here. Newer LEDs are fussier about the correct LED type, WS2811 worked on the older panels, but that was the cause for all the issues on the new panels, they needed SK6812, though WS2812 did also work. Older type LEDs still work with all three settings... go figure! Many thanks.

closed 2021-01-20 by shaunmavronicolas