FastLED/FastLED #1176: Can't get even basic functions to work with ESP32 DOIT board

mcpacific, 2021-01-18:
I'm having a hard time getting this to work on my ESP32 (DOIT DEVKIT V1). I have connected a string of WS2812B's, with the data port connected to GPIO23. My code is below, any help would be appreciated.

```
#include <FastLED.h>

// How many leds in your strip?
#define NUM_LEDS 300

// For led chips like Neopixels, which have a data line, ground, and power, you just
// need to define DATA_PIN.  For led chipsets that are SPI based (four wires - data, clock,
// ground, and power), like the LPD8806 define both DATA_PIN and CLOCK_PIN
#define DATA_PIN 23
#define CLOCK_PIN 22

// Define the array of leds
CRGB leds[NUM_LEDS];

void setup() { 
    // Serial port for debugging purposes
  Serial.begin(115200);
      // Uncomment/edit one of the following lines for your leds arrangement.
      // FastLED.addLeds<TM1803, DATA_PIN, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<TM1804, DATA_PIN, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<TM1809, DATA_PIN, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<WS2811, DATA_PIN, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<WS2812, DATA_PIN, RGB>(leds, NUM_LEDS);
      FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);
      // FastLED.addLeds<APA104, DATA_PIN, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<UCS1903, DATA_PIN, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<UCS1903B, DATA_PIN, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<GW6205, DATA_PIN, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<GW6205_400, DATA_PIN, RGB>(leds, NUM_LEDS);
      
      // FastLED.addLeds<WS2801, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<SM16716, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<LPD8806, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<P9813, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<APA102, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<DOTSTAR, RGB>(leds, NUM_LEDS);

      // FastLED.addLeds<WS2801, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<SM16716, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<LPD8806, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<P9813, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);
      // FastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN, BGR>(leds, NUM_LEDS);
      // FastLED.addLeds<DOTSTAR, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);
}

void loop() { 
  // Turn the LED on, then pause
  for(int i=0;i<NUM_LEDS;i++){
    leds[i] = CRGB::Red;
  }
  FastLED.show();
  Serial.println("Should be RED.");
  delay(500);
  for(int i=0;i<NUM_LEDS;i++){
    leds[i] = CRGB::Green;
  }
  FastLED.show();
  Serial.println("Should be GREEN.");
  delay(500);
  for(int i=0;i<NUM_LEDS;i++){
    leds[i] = CRGB::Blue;
  }
  FastLED.show();
  Serial.println("Should be BLUE.");
  delay(500);
  // Now turn the LED off, then pause
  for(int i=0;i<NUM_LEDS;i++){
    leds[i] = CRGB::Black;
  }
  FastLED.show();
  Serial.println("Should be OFF.");
  delay(500);
}
```

mcpacific, 2021-01-18:
I swapped out the Esp32 NodeMCU and even the LED strips. Turns out the erratic behavior I was observing seems to have been caused by under-voltage. I was powering the Esp32 using the micro USB connector. When I connected 5V to the VIN, it started working just fine. Has anyone else experienced this?

When I put a voltage meter on with the USB connector powered, the VIN/GND shows voltage of 4.85 volts. I suspect the voltage regulator on this board is the culprit.

marmilicious, 2021-01-19:
Glad you have it sorted out, feel free to close this ticket.  
We use github "issues" here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED

closed 2021-01-19 by mcpacific