FastLED/FastLED #1182: Jumping red LED

Aromik, 2021-01-23:
Hello!
I use an Arduino Uno. I compile and upload this sketch:
`#include "FastLED.h"
#define LED_COUNT 30
#define LED_DT A4
CRGB leds[LED_COUNT];
int hue = 0;
void setup() {
  LEDS.setBrightness(255);
  LEDS.addLeds<WS2812B, LED_DT, GRB>(leds, LED_COUNT);
  LEDS.show();
}

void loop() {
  hue -= 1;
  fill_rainbow(leds, LED_COUNT, hue);
  LEDS.show();
  delay(20);
}`
When it is running, sometimes here appears a lone red-colored LED jumping from the beginning to the end of the strip/
Where am I mistaking? Or is there a bug in fill_rainbow function?
Thank you!

5chmidti, 2021-01-23:
@Aromik take a look at #668 and the pull-request for that, #943.
From your description I'm not sure if this is the same thing you experience, but maybe it's what you mean with 'lone red-colored LED'.

Aromik, 2021-01-24:
Yes, it is. But is this problem solved now?

Aromik, 2021-01-24:
Duplicate of #668
Duplicate of #943

Aromik, 2021-01-24:
I think that this is already open. Now I'm trying to work out is it the compiler problem or the library has an issue...

Aromik, 2021-02-04:
Hi!
I don't know is this a compiler or the library problem. But I found another library that is working without any glitches: https://github.com/AlexGyver/GyverLibs/tree/master/microLED

zackees, 2025-03-19:
Fix is here:

https://github.com/FastLED/FastLED/pull/1907

zackees, 2025-03-19:
[image]

closed 2025-03-24 by zackees