FastLED/FastLED #1184: FastLED[#].showLeds() requires calling the same number of times as you have strips to work

shennerf, 2021-01-31:
I have 3 strips running off of an ESP32 running FastLED3.4.0.  I wanted the 3 strips to have different brightnesses so I am using the setup detailed in the instructions for multiple controllers.  If I use the method FastLED[#].showLeds() or the controllers[#].showLeds() options, the LEDs do not light until I call the function 3 times.  Once I call it a 3rd time, the LEDs will show.  This also works if I call the first controller, then the second, and finally the 3rd.  Once called 3 times - any controller I called will fire.  The workaround is easy - call it the same number of times you have controllers (I did test - if I only setup 2 controllers - you have to call it 2 times instead of 3).

samguyer, 2021-02-01:
@shennerf Thanks for this issue report. What you're seeing a consequence of our parallel-by-default design for the ESP32. It does exactly what you're describing: it waits for all the controllers to call show(), then sends all the bits at once. I hadn't thought about the case where you want to show just one strip. I'll need to give it some thought, but ultimately we do need to support this mode of use.

shennerf, 2021-02-01:
Thank you for looking into this.  So I assume it doesn't matter that I am only calling one controller (i.e. it checks for the count, not that all 3 are called).  Is there a more efficient way of doing this in the meantime?  I haven't seen anything negative of calling the same one 3x in a row - though it does mess me up if I'm not consistent in always calling it 3x.  I also realized that while calling FastLED.show() works fine - I HAVE to do that before starting with the individual methods or it locks up/displays garbage on the LEDs.

samguyer, 2021-02-01:
@shennerf I think you need to call show on each controller -- the code queues them up as you call each one. I think the problem with calling show on one controller is that it's waiting for the other two controllers. If you call show three times on the *same* controller, it will send the same data three times on the same output pin, which is probably what causes the garbage output.

Is the issue just about the brightness? There might be other ways to handle it besides using the global brightness.

shennerf, 2021-02-02:
Calling show 3x on one controller will display only that controller without any issues, provided you have called all of them at least once (I added an initialization to set all 3 to black).  The only time it locks up or shows bad data is when you don't first call each of them once.  I am updating OTA so can't see errors being thrown - I was guessing since that method didn't return (the LCD button stays 'depressed') that there was some uninitialized data it was hitting and running all of them once avoided the issue.

My daughter and I are doing a project -  a 2' LED ring behind frosted plexiglass for her wall - a strip around the outside, a strip on the inside, and some on the front of the ring (in a clock pattern).  That way she can choose an outside pattern, inside, and still have a 'clock' she could see. - For the morning it will emulate a sunrise and play a tune when she needs to get up.

Originally I was doing it this way so I could have 3 separate programs on each of the strings running independently on separate threads and with different brightnesses.  I can change the plan to process all 3 strands then call show on them (with different brightnesses) and should be fine.  I don't have so many LEDs that it will be a performance hit (300 total).

samguyer, 2021-02-02:
@shennerf I think you can call fadeToBlackBy to get the effect you want. Global brightness was mainly an optimization for slower microcontrollers, like the Atmel series (the global brightness scaling is folded into the process of bitbanging the data). On an ESP32, it's so fast you won't even notice. BUT I also need to fix this problem in the long run.

shennerf, 2021-02-02:
Thank you very much for your advice - I really appreciate it.

samguyer, 2022-12-07:
@polvoazul Thanks for the reminder -- this issue had dropped off my radar. I'll either add a warning, or just fix it!