FastLED/FastLED #1185: First led in series, doesn't have brightness control (esp8266, ws2812B)

Le-Juan, 2021-02-03:
The first LED on my LED strip is just always bright. I can give it a brightness of 5 and still it would be at probably 150 or even brighter...

I've used this code to test it:
```
#define FASTLED_ESP8266_RAW_PIN_ORDER
#define FASTLED_ESP8266_NODEMCU_PIN_ORDER
#include <FastLED.h>
#define DATA_PIN 14                          // hardware SPI pin SCK
#define NUM_LEDS 8
struct CRGB leds[NUM_LEDS];

void setup() {
  // put your setup code here, to run once:
  FastLED.addLeds<WS2812B, DATA_PIN>(leds, NUM_LEDS);
}

void loop() {
  // put your main code here, to run repeatedly:
  for (int i = 0; i <= 8; i++) {
    leds[i] = CHSV(random8(), 5, 5); //glow 1st led as green
  }
  FastLED.show(); // apply the function on led strip

  delay(20000);
}
```

But as of right now, no luck... is there a way to fix it? Or am I just doing something really simple wrong?

marmilicious, 2021-02-04:
> for (int i = 0; i <= 8; i++)

Your CRGB array is of length 8, leds[0] to leds[7], but you're writing data to a pixel that doesn't exist, leds[8].  No telling what wackyness that's causing in memory.  

We use github "issues" here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED

Le-Juan, 2021-02-09:
Okay, thanks! My apologies. I'll ask it on Reddit, if it doesn't work ;)

closed 2021-02-09 by Le-Juan