FastLED/FastLED #1186: After the update esp8266 to version 3.0.0-dev there was a chaotic ignition of LEDs.

whakru, 2021-02-04:
After the update https://github.com/esp8266/Arduino to version 3.0.0-dev there was a chaotic ignition of LEDs.
[image]
nodemcu, arduino 1.8.13

```
#define FASTLED_ESP8266_RAW_PIN_ORDER
#define FASTLED_INTERRUPT_RETRY_COUNT 0
#define FASTLED_ALLOW_INTERRUPTS 0

#include "FastLED.h"
#define NUM_LEDS 30
#define PIN 4
#define COLOR_ORDER RGB

CRGB leds[NUM_LEDS];
void setup() { 
  FastLED.addLeds<WS2812B, PIN>(leds, NUM_LEDS); 
  }
void loop() {
for(int x = 0; x < NUM_LEDS; x++){
leds[x] = CRGB(255,50,0);
FastLED.show();
delay(50);
leds[x] = CRGB(0,0,0);
}
}
```

It should turn on one LED at a time, but in the end it sometimes lights up the others (In random order)
I attach a video: it is slow-motion to indicate the problem.
https://youtu.be/HXVT50Wo0u4

If I remove it:
#define FASTLED_INTERRUPT_RETRY_COUNT 0
#define FASTLED_ALLOW_INTERRUPTS 0
then the situation is even worse...

whakru, 2021-02-07:
I thought the problem was the LEDs, but no.

samguyer, 2021-02-08:
@whakru Thank you for submitting this issue. We have not had a chance to look at the 3.0.0 dev environment for ESP8266, but we will. I suspect that it involves interrupts (since allowing interrupts makes it worse), but I'll have to look into it. This may be a high priority issue.

whakru, 2021-02-15:
@samguyer Do you have any suggestions on how to fix it? I don't want to stand idle.

samguyer, 2021-02-15:
@whakru That is very weird. Did it work perfectly on the pre-3.0.0 release of the dev environment? Do you have any other microcontrollers you can try? I will set up a 3.0.0 environment on my side and see if I can replicate the problem. The ESP8266 is not my primary area of expertise, but it doesn't use any fancy devices -- it's just bitbanging.

whakru, 2021-02-15:
> That is very weird. Did it work perfectly on the pre-3.0.0 release of the dev environment? Do you have any other microcontrollers you can try? I will set up a 3.0.0 environment on my side and see if [...]

OK I'll try on the old build.

erik-404err, 2021-05-18:
Im currently haveing the same issue with the [release of v3.0.0](https://github.com/esp8266/Arduino/releases/tag/3.0.0):
`
#define FASTLED_ESP8266_RAW_PIN_ORDER
#define FASTLED_INTERRUPT_RETRY_COUNT 0
#include <FastLED.h>
#define LEDPin 5
#define LEDNum 300
CRGB leds[LEDNum];

#ifndef STASSID
#define STASSID "SSID"
#define STAPSK  "PASSWORD"
#endif

void setup() {
  FastLED.addLeds<WS2812B, LEDPin, GRB>(leds, LEDNum);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
}

void loop(){
  FastLED.show();
} 
`

it causes [this](https://imgur.com/a/xU7dqQP). (not always Blue, and rarely multible LEDs at once)

cause i saw [this](https://www.reddit.com/r/FastLED/comments/ncz69g/esp32_ws2812b_starts_flickering_when_sending/l) i also tried adding a delay in the loop, but it did the same flickering on the first LED, just slower.

I could also create a new issue since my flickering is only present when connected to WiFi. On ESPCore 2.7.4 everything works fine.

kimble4, 2021-06-09:
Also seeing this issue on the release of 3.0.0 - blue, sometimes white flickering of the first LED, at a rate related to how often you call FastLED.show().

Looked for all the world like a hardware problem, with the glitching seemingly related to WiFi activity in a way that had me thinking it was power-related until I rolled ESPCore back to 2.7.4 and it became rock solid again.

kriegsman, 2021-11-02:
As of https://github.com/FastLED/FastLED/commit/ccd2c7b426179ad50e4e2c0d3f132954fc6aa72c, the problem with the ESP8266 core 3.0.x may be fixed; please give it a try and let us know if you can.