FastLED/FastLED #1190: FastLED not working on Node-MCU (esp8266-12F)

bogus105, 2021-02-13:
Guys,

why it is still not solved so one can just use this lib for so popular module like node-MCU?:/

`In file included from C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_464087\Fire2012WithPalette.ino:1:0:

C:\Users\User\Documents\Arduino\libraries\FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.008

 #    pragma message "FastLED version 3.001.008"

                     ^

In file included from C:\Users\User\Documents\Arduino\libraries\FastLED/FastLED.h:68:0,

                 from C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_464087\Fire2012WithPalette.ino:1:

C:\Users\User\Documents\Arduino\libraries\FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

In file included from C:\Users\User\Documents\Arduino\libraries\FastLED/FastLED.h:51:0,

                 from C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_464087\Fire2012WithPalette.ino:1:

C:\Users\User\Documents\Arduino\libraries\FastLED/fastpin.h: In instantiation of 'class FastPin<13u>':

C:\Users\User\Documents\Arduino\libraries\FastLED/platforms/esp/8266/clockless_esp8266.h:21:49:   required from 'class ClocklessController<13, 26, 26, 52, (EOrder)66u, 0, false, 5>'

C:\Users\User\Documents\Arduino\libraries\FastLED/chipsets.h:460:7:   required from 'class WS2811Controller800Khz<13u, (EOrder)66u>'

C:\Users\User\Documents\Arduino\libraries\FastLED/FastLED.h:111:52:   required from 'class WS2811<13u, (EOrder)66u>'

C:\Users\User\Documents\Arduino\libraries\FastLED/FastLED.h:298:39:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2811; unsigned char DATA_PIN = 13u; EOrder RGB_ORDER = (EOrder)66u]'

C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_464087\Fire2012WithPalette.ino:46:64:   required from here

C:\Users\User\Documents\Arduino\libraries\FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified

  static_assert(validpin(), "Invalid pin specified");

  ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
`

tom-miller1, 2021-02-15:
FastLED works fine on NodeMCU. Are you setting any extra `#define` statements to change ESP8266 pin order or using the standard Arduino style?

bogus105, 2021-02-15:
i had 2 issues one by one: the typedef mentioned by others, then the warning regarding i2c hardware pins. When is uploaded the sketch (i took fire2012 with pallette) it was constantly breaking. After research i found hint to disable interrupts at the beginning of the sketch. Now fire sketch is working fine but i will check if wi-fi will be working and connecting to server with interrupts switched off?
Thank you.

kriegsman, 2021-02-16:
Interrupts are only temporarily disabled for the duration of FastLED.show(), and then they’re re-enabled. WiFi should be fine; let us know what you find.

bogus105, 2021-02-16:
So good to know that! i'll check it and get back to you. Thanks!

samguyer, 2021-03-22:
@bogus105 How is it going? Did you get it working?