FastLED/FastLED #1192: Error: illegal opcode mul for mcu avr25

philibertc, 2021-02-15:
Hello, 
I'm trying to upload my FastLED program to an Attiny88 (mh-tiny88) card but I get this error:
```
/tmp/cc9qt6WO.s: Assembler messages:
/tmp/cc9qt6WO.s:4445: Error: illegal opcode mul for mcu avr25
exit status 1
Erreur de compilation pour la carte MH-ET LIVE Tiny88(16.0MHz)
```
Is there a solution to fix this?
Thanks in advance.

breviglieri, 2021-06-21:
Hi, Same problem here, Do you fix your problem?

philibertc, 2021-06-21:
nope

kriegsman, 2021-06-21:
I think the problem is that FastLED doesn’t know to recognizing that particular board as an ATtiny. It’s treating it like an ATmega, which does have the mul[tiply] instruction, while the ATtiny does not.

breviglieri, 2021-06-22:
> I think the problem is that FastLED doesn’t know to recognizing that particular board as an ATtiny. It’s treating it like an ATmega, which does have the mul[tiply] instruction, while the ATtiny does not.

I found this code on reddit to insert Attiny88 in fastled, but I still couldn't solve it.

Addition to fastpin_avr.h platform entries for FastLED to make it work with Attiny88. I added this snip right before the section for the ATMega328:

#elif defined(__AVR_ATtiny88__)

#define MAX_PIN 19
_FL_DEFPIN( 0, 0, D); _FL_DEFPIN( 1, 1, D); _FL_DEFPIN( 2, 2, D); _FL_DEFPIN( 3, 3, D);
_FL_DEFPIN( 4, 4, D); _FL_DEFPIN( 5, 5, D); _FL_DEFPIN( 6, 6, D); _FL_DEFPIN( 7, 7, D);
_FL_DEFPIN( 8, 0, B); _FL_DEFPIN( 9, 1, B); _FL_DEFPIN(10, 2, B); _FL_DEFPIN(11, 3, B);
_FL_DEFPIN(12, 4, B); _FL_DEFPIN(13, 5, B); _FL_DEFPIN(14, 6, B); _FL_DEFPIN(15, 7, B);
_FL_DEFPIN(16, 7, C); _FL_DEFPIN(17, 0, C); _FL_DEFPIN(18, 1, C); _FL_DEFPIN(19, 2, C);

#define SPI_DATA 11
#define SPI_CLOCK 13
#define SPI_SELECT 10
#define AVR_HARDWARE_SPI 1
#define HAS_HARDWARE_PIN_SUPPORT 1

Also need to add an entry for the Attiny88 platform in led_sysdefs_avr.h. Scroll down to the line with entries for the other Tiny platforms, then add another OR condition to cover the 88. Here's the full line from led_sysdefs_avr.h with my addition for the 88:

#if defined(AVR_ATmega32U2) || defined(AVR_ATmega16U2) || defined(AVR_ATmega8U2) || defined(AVR_AT90USB162) || defined(AVR_ATtiny24) || defined(AVR_ATtiny44) || defined(AVR_ATtiny84) || defined(AVR_ATtiny25) || defined(AVR_ATtiny45) || defined(AVR_ATtiny85) || defined(AVR_ATtiny88) || defined(AVR_ATtiny167) || defined(AVR_ATtiny87) || defined(AVR_ATtinyX41) || defined(AVR_ATtiny841) || defined(AVR_ATtiny441)

5chmidti, 2021-06-22:
Where another problem lies is that within FastLED there are implementations of operations done in avr assembly. It's mainly used in the implementation of math functions for avr. At other places avr specific implementations are used when the macro `__AVR__` is defined, but there are less places with assembly for that.
Files with `mul` instructions:

- hsv2rgb.cpp
- lib8tion.h
- math8.h (lib8tion)
- scale8.h (lib8tion)
- trig8.h (lib8tion)
- m0clockless.h (platforms/arm/common -> it's arm, so not important for this)

Alattice, 2022-02-28:
I had the same issue but had this error:

```
Error: illegal opcode or for mcu avr25
exit status 1
```

after comparing the default entry, I managed to get it working by changing the following:

 removing the _FL part in _FL_DEFPIN_AVR in fastpin_avr.h:

```
#elif defined(__AVR_ATtiny88__)
_IO(B); _IO(C); _IO(D);

#define MAX_PIN 19
_DEFPIN_AVR( 0, 0, D); _DEFPIN_AVR( 1, 1, D); _DEFPIN_AVR( 2, 2, D); _DEFPIN_AVR( 3, 3, D);
_DEFPIN_AVR( 4, 4, D); _DEFPIN_AVR( 5, 5, D); _DEFPIN_AVR( 6, 6, D); _DEFPIN_AVR( 7, 7, D);
_DEFPIN_AVR( 8, 0, B); _DEFPIN_AVR( 9, 1, B); _DEFPIN_AVR(10, 2, B); _DEFPIN_AVR(11, 3, B);
_DEFPIN_AVR(12, 4, B); _DEFPIN_AVR(13, 5, B); _DEFPIN_AVR(14, 6, B); _DEFPIN_AVR(15, 7, B);
_DEFPIN_AVR(16, 7, C); _DEFPIN_AVR(17, 0, C); _DEFPIN_AVR(18, 1, C); _DEFPIN_AVR(19, 2, C);

#define SPI_DATA 11
#define SPI_CLOCK 13
#define SPI_SELECT 10
#define AVR_HARDWARE_SPI 1
#define HAS_HARDWARE_PIN_SUPPORT 1
```

and keeping the "__" in the definitions for led_sysdefs_avr.h:

```
#if defined(__AVR_ATmega32U2__) || defined(__AVR_ATmega16U2__) || defined(__AVR_ATmega8U2__) || defined(__AVR_AT90USB162__) || defined(__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__) || defined(__AVR_ATtiny25__) || defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny85__) || defined(__AVR_ATtiny88__) || defined(__AVR_ATtiny167__) || defined(__AVR_ATtiny87__) || defined(__AVR_ATtinyX41__) || defined(__AVR_ATtiny841__) || defined(__AVR_ATtiny441__)
```

After these changes I was able to get FastLED working on ATtiny88. Only thing is that the pin mapping is all wonky, but poking all the pins with the LED stripe data line eventually finds it. 
If its important, I'm using FastLED 3.002.010