FastLED/FastLED #1195: Is the register keyword meaningful?

estshorter, 2021-02-27:
[The `register` keyword was deprecated in C++ 11](https://en.cppreference.com/w/cpp/language/storage_duration) and causes a warning such as the following in C++17 environment:
```
warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
```
I wonder the keyword is useful.

estshorter, 2021-04-17:
ping @kriegsman

MCUdude, 2021-09-06:
I'm also getting hundreds of warnings when compiling in a C++17 environment. It would be nice if the `register` keyword could be replaced with something else.

skorokithakis, 2021-12-11:
I am getting the same thousands of warnings.

Evelios, 2022-01-24:
I to am having this problem and with the huge increase in warnings it is significantly increasing the debug time because I have to search through a pile of warnings to find issues with my code build.

I went through the suggestions in this thread but couldn't find a temporary solution other than a library fix
https://stackoverflow.com/questions/22422741/turning-off-the-register-storage-class-specifier-is-deprecated-warning

MCUdude, 2022-01-24:
I ended up compiling with g++11 instead of g++17

```
build_unflags =
  -std=gnu++17

build_flags =
  -std=gnu++11
```

m4rc77, 2023-04-07:
If you are using [PlatformIO](https://platformio.org/) you can add in your `platformio.ini` file the following lines, in order to "silence" the warnings from FastLED until a patch is available:

```
; fix in order to get rid of 'warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]' from FastLED Library <= 3.5.0
; fix in order to get rid of 'warning: this 'while' clause does not guard... [-Wmisleading-indentation]' from FastLED Library <= 3.5.0
build_flags =
    -Wno-register
    -Wno-misleading-indentation
```
**But be warned, this will also hide similar issues in your own code!**

estshorter, 2023-05-29:
resolved: https://github.com/FastLED/FastLED/pull/1461

closed 2023-05-29 by estshorter