FastLED/FastLED #1196: Arduino IDE won't compile FastLED examples for Adafruit Matrix Portal M4 SAMD51

LEDLIT1, 2021-02-28:
My Adafruit Matrix Portal M4 SAMD51 works with their neopixel examples, but when I try FastLED I get the error code below. Can someone help me get this to work with FastLED? I have a big project and need this board to work FastLED.

`Arduino: 1.8.13 (Windows 10), TD: 1.53, Board: "Adafruit Matrix Portal M4 (SAMD51), Enabled, 120 MHz (standard), Small (-Os) (standard), 50 MHz (standard), Arduino, Off"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\jlcel\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\jlcel\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\jlcel\Documents\Arduino\libraries -fqbn=adafruit:samd:adafruit_matrixportal_m4:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -vid-pid=0000_0000 -ide-version=10813 -build-path C:\Users\jlcel\AppData\Local\Temp\arduino_build_826101 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.bossac.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.CMSIS.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\CMSIS\5.4.0 -prefs=runtime.tools.CMSIS-5.4.0.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\CMSIS\5.4.0 -prefs=runtime.tools.arduinoOTA.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4 -prefs=runtime.tools.arm-none-eabi-gcc-9-2019q4.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4 -prefs=runtime.tools.openocd.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.8.0-48-gb176eee -verbose C:\Users\jlcel\AppData\Local\Temp\arduino_modified_sketch_56285\Blink.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\jlcel\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\jlcel\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\jlcel\Documents\Arduino\libraries -fqbn=adafruit:samd:adafruit_matrixportal_m4:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -vid-pid=0000_0000 -ide-version=10813 -build-path C:\Users\jlcel\AppData\Local\Temp\arduino_build_826101 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.bossac.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.CMSIS.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\CMSIS\5.4.0 -prefs=runtime.tools.CMSIS-5.4.0.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\CMSIS\5.4.0 -prefs=runtime.tools.arduinoOTA.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4 -prefs=runtime.tools.arm-none-eabi-gcc-9-2019q4.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4 -prefs=runtime.tools.openocd.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.8.0-48-gb176eee -verbose C:\Users\jlcel\AppData\Local\Temp\arduino_modified_sketch_56285\Blink.ino

Using board 'adafruit_matrixportal_m4' from platform in folder: C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5

Using core 'arduino' from platform in folder: C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5

Detecting libraries used...

"C:\\Users\\jlcel\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"Blink.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10813 -DARDUINO_MATRIXPORTAL_M4 -DARDUINO_ARCH_SAMD -D__SAMD51J19A__ -DADAFRUIT_MATRIXPORTAL_M4_EXPRESS -DCRYSTALLESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x80C9 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Matrix Portal M4\"" "-IC:\\Users\\jlcel\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.6.5\\cores\\arduino/TinyUSB" "-IC:\\Users\\jlcel\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.6.5\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore" "-IC:\\Users\\jlcel\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.6.5\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src" -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DENABLE_CACHE -Os -DVARIANT_QSPI_BAUD_DEFAULT=50000000 -D__SAMD51J19A__ -DADAFRUIT_MATRIXPORTAL_M4_EXPRESS -DCRYSTALLESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x80C9 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Matrix Portal M4\"" "-IC:\\Users\\jlcel\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.6.5\\cores\\arduino/TinyUSB" "-IC:\\Users\\jlcel\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.6.5\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore" "-IC:\\Users\\jlcel\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.6.5\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src" -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 "-IC:\\Users\\jlcel\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS\\5.4.0/CMSIS/Core/Include/" "-IC:\\Users\\jlcel\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS\\5.4.0/CMSIS/DSP/Include/" "-IC:\\Users\\jlcel\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\jlcel\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.6.5\\cores\\arduino" "-IC:\\Users\\jlcel\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.6.5\\variants\\matrixportal_m4" "C:\\Users\\jlcel\\AppData\\Local\\Temp\\arduino_build_826101\\sketch\\Blink.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for FastLED.h: [FastLED@3.4.0]

ResolveLibrary(FastLED.h)

  -> candidates: [FastLED@3.4.0]

"C:\\Users\\jlcel\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fn [... cut ...]
C:\Users\jlcel\AppData\Local\Temp\arduino_modified_sketch_56285\Blink.ino:40:59:   required from here

C:\Users\jlcel\Documents\Arduino\libraries\FastLED\src/fastpin.h:210:24: error: static assertion failed: Invalid pin specified

  210 |  static_assert(validpin(), "Invalid pin specified");

      |                ~~~~~~~~^~

Using library FastLED at version 3.4.0 in folder: C:\Users\jlcel\Documents\Arduino\libraries\FastLED 

exit status 1

Error compiling for board Adafruit Matrix Portal M4 (SAMD51).

`

board707, 2021-03-07:
Hi, LEDLIT1
As I see, you used pin 23 for connect to LED
But maximum pin number for FastLED  in your board is 19

LEDLIT1, 2021-03-09:
What pin on the Matrix Portal M4 SAMD51 do you recommend trying? 
Below is the basic code from the FastLED Blink example. I have assigned pin 15 of the Matrix Portal M4 SAMD51. It will not compile. I am at my wits end and have a project due. I would prefer using FastLED like with all my projects, but I need some help with this one. Can anyone help?

`#include <FastLED.h>
#define NUM_LEDS 1
#define DATA_PIN 15
CRGB leds[NUM_LEDS];
void setup() { 
     FastLED.addLeds<WS2813, DATA_PIN, RGB>(leds, NUM_LEDS);
}
void loop() { 
  leds[0] = CRGB::Red;
  FastLED.show();
  delay(500);
  leds[0] = CRGB::Black;
  FastLED.show();
  delay(500);
}`

board707, 2021-03-10:
You wrote that the code not compiled. What error did you get from compiler?

LEDLIT1, 2021-03-13:
Hi board707, Here is the error message. 

``Arduino: 1.8.13 (Windows 10), TD: 1.53, Board: "Adafruit Matrix Portal M4 (SAMD51), Enabled, 120 MHz (standard), Small (-Os) (standard), 50 MHz (standard), Arduino, Off"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\jlcel\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\jlcel\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\jlcel\Documents\Arduino\libraries -fqbn=adafruit:samd:adafruit_matrixportal_m4:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -vid-pid=0000_0000 -ide-version=10813 -build-path C:\Users\jlcel\AppData\Local\Temp\arduino_build_826101 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.bossac.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.CMSIS.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\CMSIS\5.4.0 -prefs=runtime.tools.CMSIS-5.4.0.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\CMSIS\5.4.0 -prefs=runtime.tools.arduinoOTA.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4 -prefs=runtime.tools.arm-none-eabi-gcc-9-2019q4.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4 -prefs=runtime.tools.openocd.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.8.0-48-gb176eee -verbose C:\Users\jlcel\AppData\Local\Temp\arduino_modified_sketch_56285\Blink.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\jlcel\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\jlcel\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\jlcel\Documents\Arduino\libraries -fqbn=adafruit:samd:adafruit_matrixportal_m4:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -vid-pid=0000_0000 -ide-version=10813 -build-path C:\Users\jlcel\AppData\Local\Temp\arduino_build_826101 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.bossac.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.CMSIS.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\CMSIS\5.4.0 -prefs=runtime.tools.CMSIS-5.4.0.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\CMSIS\5.4.0 -prefs=runtime.tools.arduinoOTA.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4 -prefs=runtime.tools.arm-none-eabi-gcc-9-2019q4.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4 -prefs=runtime.tools.openocd.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.8.0-48-gb176eee -verbose C:\Users\jlcel\AppData\Local\Temp\arduino_modified_sketch_56285\Blink.ino

Using board 'adafruit_matrixportal_m4' from platform in folder: C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5

Using core 'arduino' from platform in folder: C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5

Detecting libraries used...

[... cut ...] sketch\Blink.ino.cpp" -o "C:\Users\jlcel\AppData\Local\Temp\arduino_build_826101\sketch\Blink.ino.cpp.o"

In file included from C:\Users\jlcel\AppData\Local\Temp\arduino_modified_sketch_56285\Blink.ino:1:

C:\Users\jlcel\Documents\Arduino\libraries\FastLED\src/FastLED.h:578:2: warning: #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info." [-Wcpp]

578 | #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info."`

kriegsman, 2021-03-15:
Hi- we just merged in a pull request that may help with this.  Try checking out the latest revision of the code and let us know if this helps.  We'll stay tuned for your report.

board707, 2021-03-15:
Hi, LEDLIT1
This is not a error. it is warning. So the code compiled, try to upload it to controller

LEDLIT1, 2021-03-16:
Hi Mr. Kreigsman,
Thank you for responding. The Fastpin.h seems to be calling out the error, but I am not sure.  Please help. Try selecting the Adafruit Matrix Portal M4 (SAMD51) in the Arduino IDE and try to compile/upload and you will see what I mean. I have asked the Adafruit forum and they can't help with FastLED. Here is the detailed procedure I have taken with the error message below.

1. Open Arduino IDE version 1.8.13
2. I confirmed the latest version of FastLED, which is FastLED 3.4.0
3. Connected my Adafruit Matrix Portal M4 (SAMD51) to my Windows 10 PC USB
4. Tools>Board>Adafruit SAMD> selected Adafruit Matrix Portal M4 (SAMD51)
5. Tools>Port> selected com37 Adafruit Matrix Portal M4 (SAMD51) is showing
6. File>Examples>FastLED> select Blink
7. In the sketch I reassign #define DATA_PIN 3 to #define DATA_PIN A0
8. Click Upload
9. Error compiling for board Adadfruit Matrix Portal M4 (SAMD51) occurs
10. Copied and pasted the error message below.
11. The Fastpin.h seems to be calling out the error, but I am not sure.
12. I have tried ALL the FastLED examples with ALL the possible pins available on multiple Matrix Portal M4 (SAMD51) and have had the same errors for 2 weeks now.

`Arduino: 1.8.13 (Windows 10), Board: "Adafruit Matrix Portal M4 (SAMD51), Enabled, 120 MHz (standard), Small (-Os) (standard), 50 MHz (standard), Arduino, Off"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\jlcel\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\jlcel\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\jlcel\Documents\Arduino\libraries -fqbn=adafruit:samd:adafruit_matrixportal_m4:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -vid-pid=239A_80C9 -ide-version=10813 -build-path C:\Users\jlcel\AppData\Local\Temp\arduino_build_439064 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.CMSIS.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\CMSIS\5.4.0 -prefs=runtime.tools.CMSIS-5.4.0.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\CMSIS\5.4.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.arduinoOTA.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4 -prefs=runtime.tools.arm-none-eabi-gcc-9-2019q4.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.openocd.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -verbose C:\Users\jlcel\AppData\Local\Temp\arduino_modified_sketch_125443\Blink.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\jlcel\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\jlcel\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\jlcel\Documents\Arduino\libraries -fqbn=adafruit:samd:adafruit_matrixportal_m4:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -vid-pid=239A_80C9 -ide-version=10813 -build-path C:\Users\jlcel\AppData\Local\Temp\arduino_build_439064 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.CMSIS.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\CMSIS\5.4.0 -prefs=runtime.tools.CMSIS-5.4.0.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\adafruit\tools\CMSIS\5.4.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.arduinoOTA.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\jlcel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.to [... cut ...]
C:\Users\jlcel\Documents\Arduino\libraries\FastLED-3.4.0\src/fastpin.h:232:71: warning: unused parameter 'val' [-Wunused-parameter]

  232 |  inline static void fastset(register port_ptr_t port, register port_t val) __attribute__ ((always_inline)) { }

      |                                                       ~~~~~~~~~~~~~~~~^~~

Using library FastLED-3.4.0 at version 3.4.0 in folder: C:\Users\jlcel\Documents\Arduino\libraries\FastLED-3.4.0 

exit status 1

Error compiling for board Adafruit Matrix Portal M4 (SAMD51).

`