FastLED/FastLED #1200: Error compiling for ATtiny 0/1

aussieW, 2021-03-05:
I am trying to compile for a ATtiny 0/1 series chip but getting errors as per below. According to #1183 it should work but doesn't for me. Any assistance in resolving would be appreciated.

Environment is:
Arduino IDE 1.8.13
Lastest version of FastLED master branch
megaTinyCore
ATtiny1614 16MHz Internal Clock

```
Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "ATtiny1614/1604/814/804/414/404/214/204, No (saves space), ATtiny1614, 16 MHz Internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), UPDI (pick this unless you have an HV UPDI programmer!), 8ms, Closer to 5v"

In file included from C:\Users\austi\Documents\Arduino\libraries\FastLED\examples\DemoReel100\DemoReel100.ino:1:0:

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/FastLED.h:585:2: warning: #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info." [-Wcpp]

 #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info."

  ^~~~~~~

In file included from C:\Users\austi\Documents\Arduino\libraries\FastLED\examples\DemoReel100\DemoReel100.ino:1:0:

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Users\austi\Documents\Arduino\libraries\FastLED\src/platforms/avr/fastled_avr.h:4:0,

                 from C:\Users\austi\Documents\Arduino\libraries\FastLED\src/platforms.h:41,

                 from C:\Users\austi\Documents\Arduino\libraries\FastLED\src/FastLED.h:52,

                 from C:\Users\austi\Documents\Arduino\libraries\FastLED\examples\DemoReel100\DemoReel100.ino:1:

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/platforms/avr/fastpin_avr.h: In static member function 'static volatile uint8_t& __gen_struct_DDRA::r()':

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/platforms/avr/fastpin_avr.h:59:26: error: 'DDRA' was not declared in this scope

 #define _FL_IO(L,C) _RD8(DDR ## L); _RD8(PORT ## L); _RD8(PIN ## L); _FL_DEFINE_PORT3(L, C, _R(PORT ## L));

                          ^

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/platforms/avr/fastpin_avr.h:50:79: note: in definition of macro '_RD8'

 #define _RD8(T) struct __gen_struct_ ## T { static inline reg8_t r() { return T; }};

                                                                               ^

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/platforms/avr/fastpin_avr.h:65:3: note: in expansion of macro '_FL_IO'

   _FL_IO(A,0)

   ^~~~~~

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/platforms/avr/fastpin_avr.h:59:26: note: suggested alternative: 'DD0'

 #define _FL_IO(L,C) _RD8(DDR ## L); _RD8(PORT ## L); _RD8(PIN ## L); _FL_DEFINE_PORT3(L, C, _R(PORT ## L));

                          ^

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/platforms/avr/fastpin_avr.h:50:79: note: in definition of macro '_RD8'

 #define _RD8(T) struct __gen_struct_ ## T { static inline reg8_t r() { return T; }};

                                                                               ^

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/platforms/avr/fastpin_avr.h:65:3: note: in expansion of macro '_FL_IO'

   _FL_IO(A,0)
[... 170 lines cut ...]

                 from C:\Users\austi\Documents\Arduino\libraries\FastLED\examples\DemoReel100\DemoReel100.ino:1:

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/fastpin.h: In instantiation of 'class FastPin<3>':

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/platforms/avr/clockless_trinket.h:100:49:   required from 'class ClocklessController<3, 6, 8, 6, (EOrder)66, 0, false, 10>'

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/chipsets.h:477:7:   required from 'class WS2811Controller800Khz<3, (EOrder)66>'

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/FastLED.h:111:52:   required from 'class WS2811<3, (EOrder)66>'

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/FastLED.h:302:39:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2811; unsigned char DATA_PIN = 3; EOrder RGB_ORDER = (EOrder)66]'

C:\Users\austi\Documents\Arduino\libraries\FastLED\examples\DemoReel100\DemoReel100.ino:32:64:   required from here

C:\Users\austi\Documents\Arduino\libraries\FastLED\src/fastpin.h:210:2: error: static assertion failed: Invalid pin specified

  static_assert(validpin(), "Invalid pin specified");

  ^~~~~~~~~~~~~

exit status 1

Error compiling for board ATtiny1614/1604/814/804/414/404/214/204.
```

5chmidti, 2021-03-06:
That pull-request has not been merged yet. If you are in a rush, you could use the branch where the pull-request is from ([jasoncoon:master](https://github.com/jasoncoon/FastLED)), otherwise you'll have to wait.

aussieW, 2021-03-06:
Thankyou. I tried out Jason's version and it compiles successfully.

closed 2021-03-06 by aussieW

Hely0n, 2022-04-13:
@aussieW Did uploading it work for you too? I'm also using this branch and compiling works but all LEDs are white. I know this comes from a wrong CPU clock, as I had the same error with an ATtiny85 last week. But the fix I used there (build_flags = -DCLOCK_SOURCE=0) doesn't work with the Attiny1604. OSCCFG Fuse is set to 0b1, so it should work in 16MHz mode.